Atmosphere-NX/Atmosphere #1451: 0.19.1 boot issue

myeuropeanPAL, 2021-04-16:
While booting into EmuNAND (not emuMMC) with 0.19.1 I got a black screen after the atmosphere logo, the nintendo switch logo never appears.
Booting into CFW without emuNAND seems wo be working fine.
This is not an issue with 0.19.0
Using the latest sig patches for 0.19.1

After removing all folders from atmosphere\contents it is booting with 0.19.1 as well
Launched with hekate 5.5.5
Switch firmware is 11.0.1
EmuNAND firmware is 12.0.0

No crash reports, just a black screen.

SciresM, 2021-04-16:
Re-open using the issue templates.

closed 2021-04-16 by SciresM

myeuropeanPAL, 2021-04-16:
I re-opened it using the template, but it got closed again.
https://github.com/Atmosphere-NX/Atmosphere/issues/1452

fincs, 2021-04-16:
https://github.com/Atmosphere-NX/Atmosphere/issues/1452#issuecomment-821564714