Atmosphere-NX/Atmosphere #1452: 0.19.1 boot issue - second try

myeuropeanPAL, 2021-04-16:
## Bug Report

### What's the issue you encountered?

While booting into EmuNAND (not emuMMC) with 0.19.1 I got a black screen after the atmosphere logo, the nintendo switch logo never appears.
Booting into CFW without emuNAND seems wo be working fine.
This is not an issue with 0.19.0

After removing all folders from atmosphere\contents it is booting with 0.19.1 as well

### How can the issue be reproduced?

By having sysmodule folders in atmosphere\contents and booting into EmuNAND

### Crash Report

No crash reports, just a black screen.

### System Firmware Version

Switch firmware is 11.0.1
EmuNAND firmware is 12.0.0 (updated from 11.0.1 where the issue was the same)

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: 5.5.5
- Do you have additional kips or sysmodules you're loading: sig patches for 0.19.1
- Homebrew software installed: Tesla, sys-botbase

### Additional context?

- Additional info about your environment: N/A

masagrator, 2021-04-16:
> sig patches for 0.19.1

Piracy is not supported.

> SD card is exFAT formatted

Don't use exFAT.

fincs, 2021-04-16:
Piracy is not supported.

closed 2021-04-16 by fincs