Atmosphere-NX/Atmosphere #1453: ChoiDuJourNX crashes

ryotoko, 2021-04-18:
## Bug Report

### What's the issue you encountered?

ChoiDuJourNX 1.0.2 crashes the system when attempting to load it from the HBmenu.
Games and other homebrew work fine.

### How can the issue be reproduced?

I am attempting to update my emuMMC FW, in the same way I always do, which is:
1) Attain the newest versions of Hekate, Atmosphere, and FW files
2) Unzip to SD
3) Boot into new version Atmosphere emuMMC
4) Update emuMMC using ChoiDuJourNX from HBmenu
5) ... except it crashes. Every time.

### Crash Report

https://gist.github.com/ryotoko/9636c3e4cdba296484adb67e9d4e5a8c

### System Firmware Version

emuMMC is 11.0.1
system is 12.0.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate 5.5.5
- Official release or unofficial build: Official 0.19.1
- Do you have additional kips or sysmodules you're loading: Sigpatches linked here:
- https://gbatemp.net/threads/sigpatches-for-atmosphere-hekate-fss0-fusee-secondary-only.571543/
- Homebrew software installed: ChoiDuJourNX, Tinleaf, nxmtp

### Additional context?

- Additional info about your environment:
Drawing a blank, I've done this same process a dozen times before with no problems.
I have tried with and without mesosphere packages.
Tried adding fullsvcperm=1 to hekate_ipl.ini emuMMC launch profile, nothing.
[image]

carcaschoi, 2021-04-18:
use daybreak instead

SciresM, 2021-04-18:
Certain homebrew (from prior to 2019) needs to be recompiled to support the sm changes made in official firmware 12.0.0/atmosphere 0.19.1.

wait for an update to be released for the homebrew in question, or use daybreak instead.

closed 2021-04-18 by SciresM

IvanVentur, 2021-04-29:
same issue reproduced on Atm 0.19.1 and system is 10.X.X

fincs, 2021-04-29:
As stated above, use Daybreak.