Atmosphere-NX/Atmosphere #1454: Fatal Error

azznpride, 2021-04-18:
## Bug Report

### What's the issue you encountered?
Fatal Error trying to boot to atmosphere
see picture
[image]

[ Describe the issue in detail and what you were doing beforehand. ]</br>
Just trying to launch atmosphere after getting all the required files since i haven't update 
[ Did you make any changes related to Atmosphère itself? ]</br>
no

### How can the issue be reproduced?
Inject hekate v5.5.5 & Nyx v1.0.2
Launch fuslee

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[report_000000000792c7ca.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6330355/report_000000000792c7ca.zip)

### System Firmware Version
12.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- hekate
- Official release or unofficial build: 
  - Official release hekate v5.5.5 & Nyx v1.0.2
- Do you have additional kips or sysmodules you're loading:
- no
- Homebrew software installed: [ * ]
-no
### Additional context?
no

azznpride, 2021-04-18:
Just notice that i upload the wrong one for Crash report not sure it matter but here the matching one in the picture
[report_00000000078f0c08.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6330374/report_00000000078f0c08.zip)

SciresM, 2021-04-18:
The automatic cleanup routines for migrating from atmosphere 0.18.x -> 0.19.x are failing (the stuff that makes stratosphere.romfs not fail, since we used to distribute loose contents).

Result "DirectoryNotEmpty" makes approximately no sense, I'm going to tentatively presume there's an archive bit issue or something on one of the folders.

In any event, just delete the `/atmosphere/contents` folder wholesale, and maybe also run hekate's archive bit fixer. Your issue should be resolved, after that.

closed 2021-04-18 by SciresM