Atmosphere-NX/Atmosphere #1461: Fatal Error program ID 010000000000002b

Tidybloke, 2021-04-21:
## Bug Report

I have searched for days and tried varies fixes with the same error code but I understand the error codes for one persons problem may not necessarily be the same for others.</br>
</br>

### What's the issue you encountered?

I updated my Switch to firmware 12.0.0 and updated Atmosphere and used the latest hekate_ctcaer. I get past a couple of Atmosphere splash screens then the following error appears</br>

A fatal error occured when running Atmosphere</br>
Program ID: 010000000000002b</br>
Error Desc: std::abort() called (0xffe)</br>

I then used the latest Fusee-primary payload provided on your Github and received the same error.</br>

### How can the issue be reproduced?

As per Description above</br>

### Crash Report

[report_00000000315e4b3c.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6353801/report_00000000315e4b3c.zip)

### System Firmware Version

12.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:  Fusee-Primary
- Official release or unofficial build: 
  - [ Official release version 0.17.0]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ N/A ]
     - [N/A]

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:  hekate_ctcaer
- Official release or unofficial build: 
  - [ Official release version 5.5.5]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ N/A ]
     - [N/A]

- Do you have additional kips or sysmodules you're loading:    No

- Homebrew software installed: [ * ]

> - Checkpoint

### Additional context?

- Additional info about your environment:
- [N/A ]

uCZMG, 2021-04-21:
Update to Atmosphere 0.19.1, 0.17.0 isn't compatible with 12.0.0.

(Also the latest version of Checkpoint isn't compatible yet, but there is a fork that solves the issue: https://github.com/SciresM/Checkpoint/releases/tag/ams-19-compat)

Tidybloke, 2021-04-21:
Sorry maybe my bad but the 0.17.0 I quoted was for the Fusee-Primary. The Atmosphere Version I am trying to use is 0.19.1.

I was also meant to mention in my initial report is that following the Atmosphere Update and that not working I renamed the Atmosphere Folder on the SD Card and then dragged a clean version of Atmosphere over with the same boot Error unfortunately.

SciresM, 2021-04-21:
You have a corrupt PRODINFO/system settings save.

Follow the instructions/directions in #1256 for guidance. You'll need to restore a non-corrupt backup and/or delete the system settings savedata.

closed 2021-04-21 by SciresM