Atmosphere-NX/Atmosphere #1468: Switch Won't Boot Into Latest CFW

cDOTwav, 2021-04-27:
## Bug Report

### What's the issue you encountered?

I've installed both hekate version 5.5.5 and Atmosphere version 0.19.1 and now my Switch won't boot into CFW, my Switch is on firmware 11.0.1, which probably explains why it's doing that, but even so I have no clue on how to install the latest firmware since it won't boot.

### How can the issue be reproduced?

Be on any version lower than 12.0.1 with the latest Atmosphere version (as of now 0.19.1), and try to boot into CFW.

### Crash Report

N/A

### System Firmware Version

11.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: Official release version 5.5.5
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: N/A

### Additional context?

- Additional info about your environment: N/A

SciresM, 2021-04-27:
I can't reproduce this locally. Screenshot of your atmosphere and atmosphere/contents directory, please?

cDOTwav, 2021-04-27:
Okay.
[image]
[image]

SciresM, 2021-04-27:
You said you have no custom sysmodules.

In fact, you have 7 and a custom theme.

I can't help you if you lie to me.

In any event, delete your custom sysmodules/theme. Your issue will be solved.

closed 2021-04-27 by SciresM

cDOTwav, 2021-04-27:
I didn't even know what that meant, don't accuse me of lying when I'm just looking for help, there's no reason for me to lie. You could've just pointed out that I did instead of accusing me of that shit. Anyway thanks, I guess, or could you be lying about the solution? 🤔

Masamune3210, 2021-04-27:
If you dont know what something is, instead of saying no, why not just ask? If you don't know what you have you probably shouldn't be messing with cfw much anyway. Do some research

Deero27, 2021-07-25:
> You said you have no custom sysmodules.
> 
> In fact, you have 7 and a custom theme.
> 
> I can't help you if you lie to me.
> 
> In any event, delete your custom sysmodules/theme. Your issue will be solved.

just wanted you to know, this actually helped me, thanks 👍

nullnash, 2022-01-28:
@SciresM @Masamune3210 Wow you two are jerks people looking for help and you act all high and mighty. Bunch of d-bags right here.

takoyaki99, 2022-01-28:
The issue tracker isn’t for help. Go to discord for that. They’re providing a service by making and maintaining Atmosphere…for free…

Masamune3210, 2022-01-28:
Dude, we weren't acting like D-bags, if you don't understand what is asked, just say that, don't give false information. Especially when asking for help for something you get for free