Atmosphere-NX/Atmosphere #1469: Default hosts file in 0.19.1 breaks WiFi connectivity check

dantho281, 2021-04-27:
## Bug Report

### What's the issue you encountered?

When using 0.19.1, the default hostsfile for dns_mitm has been updated to include `127.0.0.1 *nintendo*`, which appears to be breaking connectivity checks within the Switch's OS, preventing WiFi from connecting again.

### How can the issue be reproduced?

1. Update to Atmosphere 0.19.1
2. Enable dns_mitm
3. Attempt to connect to WiFi, will fail with the error that the server did not respond.
4. Unload dns_mitm
5. WiFi connection succeeds

### System Firmware Version

12.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
fusee latest launching.
Standard atmosphere used from `latest` github tag release.

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Having looked through the code, I'm unsure as to where this default hosts list is coming from.
It appears only 
```
# Nintendo telemetry servers
127.0.0.1 receive-%.dg.srv.nintendo.net receive-%.er.srv.nintendo.net
```

should be added, as per the docs, however mine included the `*nintendo*` line as above.
Is it possible an older version made an incorrect hosts list, that has then persisted in upgrade?

SciresM, 2021-04-27:
You got a custom hosts file from someone else.

Atmosphere has never used a hosts file that simplified, nor will it, specifically because of the issue you mention.

SciresM, 2021-04-27:
(The file inside the zips in the release does not contain that line.)

closed 2021-04-27 by SciresM

dantho281, 2021-04-27:
You are, of course, totally correct.
I was doubting myself because I literally went and looked at `dnsmitm_host_redirection.cpp` to pick out how this could happen, and it clearly couldn't.

What happened is at some point, I downloaded something called `JustInsertTheStuff` which included this hosts file.
Seems this was many months ago, I've no clue where I got it from or why, because I normally update properly.
Also confused why it didn't break earlier, if I'm very honest. But that's very much by the by.

I also know this isn't on you to fix, but I'm wondering if there's a way to try and detect such faults, and have Atmosphere warn about it. But that'd be something I'd have to look into, as I say, not on you to fix someone else's screwup.

But anyway - a general warning to check for old things on an SD, because clearly someone didn't know what they were doing, and someone else (me) tried to save some time at some point.

carcaschoi, 2021-04-28:
> You are, of course, totally correct.
> I was doubting myself because I literally went and looked at `dnsmitm_host_redirection.cpp` to pick out how this could happen, and it clearly couldn't.
> 
> What happened is at some point, I downloaded something called `JustInsertTheStuff` which included this hosts file.
> Seems this was many months ago, I've no clue where I got it from or why, because I normally update properly.
> Also confused why it didn't break earlier, if I'm very honest. But that's very much by the by.
> 
> I also know this isn't on you to fix, but I'm wondering if there's a way to try and detect such faults, and have Atmosphere warn about it. But that'd be something I'd have to look into, as I say, not on you to fix someone else's screwup.
> 
> But anyway - a general warning to check for old things on an SD, because clearly someone didn't know what they were doing, and someone else (me) tried to save some time at some point.
> 
> Thank you for your time, appreciate you taking the time to confirm my own doubts when I was in the wrong on this one.

You have to use this host file with [nifm_ctest] exefs_patches to Skips connection test to allow connections to networks without internet access or dns wildcard blocks. This host file block all Nintendo connection include testing connection website