Atmosphere-NX/Atmosphere #1472: Incompatibility with older emuNAND

apassiou, 2021-04-29:
## Bug Report

### What's the issue you encountered?

Created emuNAND using Hekate 5.5.5 (latest as of today) off of sysNAND 4.1.0.
SD Card is FAT32 formatted and clean.

Vanilla Atmosphere (0.19.1) will not boot into this low of a version emuNAND and results in a black screen after the Atmosphere logo. Both File based and Partition based emuNAND is affected.

My suspicion is that an ini/config file is missing entries for such early fw builds. Not sure at what point compatibility breaks (what the earliest possible emuNAND version is).

Temporary solution is to use to Kefir load up Atmosphere, it has the necessary entries. Boot into the emuNAND and update it with Sunrise. After that vanilla Atmosphere will boot successfully.

Atmosphere packages that do NOT work: Deepsea, SDsetup, Kosmos. Those are the ones I tried, all lead to black screen. Kefir is the only one that works.

[ Did you make any changes related to Atmosphère itself? ]</br>
No

[ If so, make sure to include details relating to what exactly you changed. ]</br>
N/A

### How can the issue be reproduced?

Create an emuNAND off of sysNAND on version 4.1.0 and attempt to run Atmosphere. Possibly other versions are affected.

### Crash Report

N/A

### System Firmware Version

4.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee-primary.bin or through Hekate
- Official release or unofficial build: 
   Official version 0.19.1
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Nothing, Vanilla Atmosphere

### Additional context?

- Additional info about your environment:
Vanilla Atmosphere loaded through fusee-primary.bin , no layerfs, no patches, no homebrew.

SciresM, 2021-04-29:
>4.1.0

duplicate of #1460, already resolved in next release.

Use the WITHOUT_MESOSPHERE release until then.

closed 2021-04-29 by SciresM

apassiou, 2021-04-29:
> > 4.1.0
> 
> duplicate of #1460, already resolved in next release.
> 
> Use the WITHOUT_MESOSPHERE release until then.

Cool thanks!