Atmosphere-NX/Atmosphere #1474: Black Screen at start after update to 12.0.1 (Atmosphere 0.19.1)

AlejandroMHOB, 2021-04-29:
What's the issue you encountered?
After updating Atmosphere to version 0.19.1, also updating the Hetake bootloader to version 5.5.5 (All official releases), I proceed to update the console to the latest version 12.0.1. Right after updating, when it restarts, the atmosphere logo appears and the screen goes black, having to force a hard reset to turn it off.

As I could read on other sites, atmosphere 0.19.1 is compatible with 12.0.1, but updating it stopped working.

How can the issue be reproduced?
--------------------------------------
From the Hekate 5.5.5 Bootloader, both starting from the Emunand main button or from the Payload section with the fusee-primary (original from 0.19.1), the result is the same; after the Atmosphere logo, the screen goes black and does not respond except hard reset. 

Crash Report
--------------
Atmosphere cannot start, so it does not generate any kind of Feedback in the corresponding folders of the SD. 

System Firmware Version
---------------------------
OFW 12.0.1 / CFW Atmosphere 0.19.1 (Official release)

Environment?
---------------
Bootloader hekate v5.5.5 & Nyx v1.0.2 (Official release)

Goldleaf 0.9
Sys-clk 1.0.1
NXthemesInstaller 2.6.1
JKSV 06.0.7.2020

Before opening the ticket, I have updated all the programs to their latest version and then tried to eliminate the different homebrew programs to see that they did not conflict due to incompatibility with new versions. 

Additional context?
---------------------
Console updated directly from 11.0.1, both OFW and CFW .

SciresM, 2021-04-29:
Does atmosphere boot with all custom system modules removed?

AlejandroMHOB, 2021-04-29:
No, same result, a black screen after Atmosphere's Logo :S

SciresM, 2021-04-29:
Please post a screenshot of your /atmosphere directory, and of your /atmosphere/contents directory.

AlejandroMHOB, 2021-04-29:
There you have it. :)

[image]
[image]

SciresM, 2021-04-29:
You did not remove all custom system modules. I see three and a custom theme.

Remove them, it will solve your problem.

closed 2021-04-29 by SciresM

AlejandroMHOB, 2021-04-29:
Please, can you tell me what I should delete specifically?

SciresM, 2021-04-29:
Just delete everything inside /atmosphere/contents.

That'll delete the three custom sysmodules, the custom theme, and the two game mods you have.

You can add stuff back individually afterwards to determine what's the problem.

AlejandroMHOB, 2021-04-29:
I deleted all the content from the Atmosphere / Contents folder.
Unfortunately that did not fix my problem. I keep seeing a black screen after the Atmosphere Logo.

SciresM, 2021-04-29:
I am *very* inclined to believe this is a user error. Contact SciresM#0524 on discord, and note that this is *not* a help forum, it's a bug reporting forum.