Atmosphere-NX/Atmosphere #1481: The Software was closed because an error occurred

NaturalHarmonia, 2021-05-02:
## Bug Report

### What's the issue you encountered?

When playing Monster Hunter Rise I often get the "The software was closed because an error occurred" error since I updated my Switch firmware to 12.0.1 and Monster Hunter Rise to 2.0.0. The only changes I've made to Atmosphere was updating to 0.19.2.

### How can the issue be reproduced?

The issue can be reproduced just by playing the game naturally or by going to the weapons shop and creating decorations and then exiting and/or trying to set the decoration. The decorations I usually try to create are Speed Sharpening and Dragon Attack.

### Crash Report

https://www.mediafire.com/file/9ldg4ke63j5h2bz/08368065-35f5-c542-43dc-91f4b4096253/file

### System Firmware Version

System firmware is 12.0.1.

### Environment?

Atmosphere was launched by fusee
The Atmosphere release I used was official.
Kip patches? Yes I do believe I have additional kip_patches.
I have some apps like Edizon, Goldleaf, and etc. installed.

### Additional context?

N/A

SciresM, 2021-05-02:
Do you have custom system modules? Please provide a screenshot of /atmosphere/contents.

The error is in account code, which atmosphere doesn't interact with in any way that I'm aware of.

NaturalHarmonia, 2021-05-02:
[image]
[image]
[image]

I have cheat codes for the game but I had all cheats turned off when this was happening.

SciresM, 2021-05-02:
Looks like you have tesla installed. Does the error occur without tesla? 

Please delete /atmosphere/contents/42....7E51A folder and try again, uploading the report if you can.

NaturalHarmonia, 2021-05-02:
Looks like it still occurs even without Tesla since I got rid of that folder and the error occurred when I was going through my equipment. 

https://www.mediafire.com/folder/nnhbfpxd9x43v/erpt_reports

SciresM, 2021-05-02:
...what happens when you try to access the eShop?

NaturalHarmonia, 2021-05-02:
> ...what happens when you try to access the eShop?

I'm banned from Eshop.

thecrimsonsword, 2021-05-02:
> I'm banned from Eshop and have been way before this happened.

So basically you're doing piracy

SciresM, 2021-05-02:
Yeah, that's the problem.

Fwiw, here's what's happening:

* Account reaches out to try to connect to dauth
* Dauth returns "account banned" error code
* Account translates that into 2124-4508, and triggers the erpt report you sent.

Dunno if this is new logic in 12.0.0 or whatever, but I'm fairly confident it's not an actually an atmosphere problem.

Perhaps try playing in airplane mode or whatever.

SciresM, 2021-05-02:
If you want to try to resolve your issue further, you can also try seeing if a crash report (/atmosphere/crash_reports) is generated if you edit exosphere.ini to disable usermode exception handlers. 

But yeah, I'm as confident as I can be that this isn't an ams issue.

closed 2021-05-02 by SciresM

NaturalHarmonia, 2021-05-02:
> If you want to try to resolve your issue further, you can also try seeing if a crash report (/atmosphere/crash_reports) is generated if you edit exosphere.ini to disable usermode exception handlers.
> 
> But yeah, I'm as confident as I can be that this isn't an ams issue.

The error still occurred even with Airplane mode on. Key: disable_user_exception_handlers, default: 0. That's what it says in my exopshere.ini file. I think it's already disabled.

SciresM, 2021-05-02:
No, 0 = "not disabled".

Change the value to 1, to disable user exception handlers.

NaturalHarmonia, 2021-05-02:
> No, 0 = "not disabled".
> 
> Change the value to 1, to disable user exception handlers.

Oh and by the way my SD card is Exfat and I've heard that Exfat is known to corrupt. Do you think that could be the cause for the problem?

NaturalHarmonia, 2021-05-02:
> No, 0 = "not disabled".
> 
> Change the value to 1, to disable user exception handlers.

Here are the crash reports.

https://www.mediafire.com/folder/cqhxl5r41apbu/crash_reports

SciresM, 2021-05-02:
I'll perform a cursory check, but "game data abort" really isn't a thing that atmosphere can cause without weird stuff.

NaturalHarmonia, 2021-05-02:
> I'll perform a cursory check, but "game data abort" really isn't a thing that atmosphere can cause without weird stuff.

Alright not sure what any of that is but I'll just wait.

NaturalHarmonia, 2021-05-02:
Okay I think I have found what is causing the problems. It's the white helmet talked about in this gbatemp thread that I used to use cheat codes. It's crashing my game but there's a way to remove it apparently.

https://gbatemp.net/threads/monster-hunter-rise-cheats-gear-builds-discussion.585750/