Atmosphere-NX/Atmosphere #1482: Some dmnt limitation cause a large cheats not all loaded in EdiZon

deckyfx, 2021-05-02:
## Bug Report

### What's the issue you encountered?

some dmnt limitation cause a large cheats not all loaded

### How can the issue be reproduced?

load a huge amount of cheats

### Crash Report

Not a crash bug

### System Firmware Version

Unrelated i think

### Environment?

hekate 5.5.5
Atmosphère 0.19.2
Edizon snapshot

### Additional context?
The issue is mentioned in [this](https://github.com/WerWolv/EdiZon/issues/163) issue, 

It seems there are maximum lines this plugin can load, when I put a lot of codes, like the example below, some codes is not loaded to EdiZon. and as the reply pointed out, seems the problem is from atmosphere dmnt limitation

SciresM, 2021-05-02:
Correct. This is intended behavior, and not a bug.

closed 2021-05-02 by SciresM

SciresM, 2021-05-02:
Looking at the cheat you linked...rewrite it to use a loop, man.

Loops are supported.

deckyfx, 2021-05-03:
[](url)

> Looking at the cheat you linked...rewrite it to use a loop, man.
> 
> Loops are supported.

Thank you for your reply

That's is also what I think, any info for the code format?