Atmosphere-NX/Atmosphere #1483: Nintendo Switch sysfw crashes on boot with or without cfw after months of no usage.

Terraeris, 2021-05-05:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

### What's the issue you encountered?

Launched the system in stock after months of using only emummc, and got a crash on boot. Tried to then launch in syscfw in case I messed up the fuses by accident and got an error with atmosphere.
No changes to atmosphere, I normally use Hekate, but for the test as to why it's giving an error, I used a different SD formatted and using the latest atmosphere download (0.19.2).

### How can the issue be reproduced?
Launching my system normally

### Crash Report

https://gist.github.com/Terraeris/7c9a641edad69e49cd588be6361090c2

### System Firmware Version

11.0.1 sysfw, 12.0.1 cfw(emummc)</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- fusee-primary.bin from the latest release.
- Official release or unofficial build: 
  - Official
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Stock atmosphere, but I can give a list of homebrew/kips/sysmodules on my other SD used for launching atmosphere through hekate if that would help.

### Additional context?

- Additional info about your environment:
- EmuMMC usually updated through Daybreak

ELY3M, 2021-05-05:
Do you have game cart inserted?    try to boot without game cart inserted.

SciresM, 2021-05-05:
Post a screenshot of /atmosphere/contents.

Terraeris, 2021-05-05:
> Do you have game cart inserted? try to boot without game cart inserted.

Ah that was the issue. I'm just an idiot lmao, thanks. Don't know why I didn't think of that.
It works fine now that I've updated the sysmmc.

closed 2021-05-05 by Terraeris

ELY3M, 2021-05-05:
I was right :)    glad I helped you.