Atmosphere-NX/Atmosphere #1485: Game running panic occurred. Title ID: 010000000000002B

Nowheremv, 2021-05-05:
## Bug Report
Panic occurred while running atmosphere
Title ID: 010000000000002B
Error: std: : abort(0xFFE)
Report saved to /atmosphere/fatal_errors/report_35f66145f.bin
[ * ]</br>
Start playing Persona Strikers

### What's the issue you encountered?
Run Persona Strikers

### How can the issue be reproduced?
I don't know

### Crash Report

[report_35f66145f.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6428812/report_35f66145f.zip)

### System Firmware Version

12.0.0 | AMS 0.19.2 | E

### Environment?

- What bootloader fusèe and hekate
- Official release 0.19.2 download via github
- Do you have additional kips or sysmodules you're loading: i dont know
- Homebrew : 3.4.0

### Additional context?

- I've played the other games just fine: MH rise, immortals fenyx rising and curse of dead god. All perfect. Only Persona Strikers got problem
- Please watch my vid about the error: https://imgur.com/bzoHDGV

SciresM, 2021-05-05:
Are you using file-based emummc?

Nowheremv, 2021-05-06:
> Are you using file-based emummc?

Hi SciresM , i'm so sorry for late reply. I'm using file-based emummc which was create by hekate.

Nowheremv, 2021-05-08:
Sloved: 
To anyone who run into this problem: re-create emuMMC via sd partion, dont use sd file

SciresM, 2021-05-08:
That would solve the problem, yes, I think the problem will also be solved by commits I made the other day in the next atmosphere release.

closed 2021-05-08 by SciresM