Atmosphere-NX/Atmosphere #1486: Invalid file in Atmosphere folder

peruserprecurer, 2021-05-06:
## Bug Report

### What's the issue you encountered?

After instability in my system with random crashes when using homebrew like AIO-updater and being unable to open any game with it being labeled as corrupt (all games I've tried are stored on my SD card as my Switch refuses to read my game card in CFW), I decided to replace all files in the Atmosphere folder with fresh ones. As such, I cut the files to my computer, but one couldn't be transferred. It was this file in the error report directory:
[image]
As it contains invalid characters, I can't delete it, and it (presumably) caused continued instability when launching emuMMC as Atmosphere immediately crashed with a 2002-3502 error when launching with the fresh files.

### How can the issue be reproduced?

N/A (I have no idea)

### Crash Report

https://gist.github.com/peruserprecurer/30d08ca2e5598b9cf4fb2845cb3b431e

### System Firmware Version

11.0.1

### Environment?

The above crash report was created when launching with Hekate (5.5.5) on Atmosphere 0.19.2. Launching with Fusee created no crash report at first and then booted me into CFW with the above-mentioned problems (not being able to launch games, et c.)
No sysmodules are used as Atmosphere is fresh.

### Additional context?

I have no way to delete this file as Windows refuses to have anything to do with it, as pictured:
[image]
(the error says that the name is too long or invalid.)
Basically, I'm stuck up Shit Creek with no paddle, probably because some file was written incorrectly. If you have some solution to this problem that I couldn't find on this World Wide Web, please tell me.
Also, Hekate sometimes claims that my SD card is in 1-bit mode sometimes or is grouchy with Nyx, saying that it's the wrong version. I won't harp on about that, though.
This was what happened when I booted into CFW that second time: https://mega.nz/file/QjoFEKzT#Pk5O84aBBlRochA5g3eNtiytzJMB1wxZy29G7ROtnrY

peruserprecurer, 2021-05-06:
That's a good point. I'm using FAT32 and my SD card is legit and only around a year old, so I'd hope it isn't failing. I ran it in h2testw and it looked to be perfectly fine. I do have a second microSD card that I intended to use ended up using in my phone instead, and it seems to have been needed as it's almost full right now. Here's a pic of the card:
[image]
And here's how full it is:
[image]

peruserprecurer, 2021-05-07:
*Sigh* I guess I'll just have to bite this bullet, then. Do you have any idea why this file came to be in the first place? Cheers.

SciresM, 2021-05-07:
Best guess is forced shutdown during FS writing a file back to disk, and the backup FAT failing.

Not really something ams has any power to help with, tbh...

closed 2021-05-07 by SciresM