Atmosphere-NX/Atmosphere #1487: Daybreak will fail to validate Updates with Mac OS ._ files present

reneroth, 2021-05-07:
### What's the issue you encountered?

Daybreak will throw an error of "Failed to get update information, Result: 0x00234a02" when selecting a valid update folder if Mac OS metadata files (`._filename`) are present.

### How can the issue be reproduced?

Copy update folder from a Mac OS HFS+ partition to an ExFat/FAT32 SD card, try to run update.

### Crash Report

N/A

### System Firmware Version

10.0.4

### Environment?

- booted with fusee
- Atmosphere 0.19.2 with Mesosphere

### Additional context?

Not sure if altering the validation routine to ignore those kind of meta files is feasible or might even be dangerous - maybe the error message could be amended to provide information that unexpected additional files were detected?
Maybe it could be documented somewhere for Mac users to run `dot_clean /Volumes/SDCARD` after copying, or at least to be on the lookout for those kind of hidden junk files.

masagrator, 2021-05-07:
Add Nintendo Switch to MacOS exception list so it won't index files on sdcard.

SciresM, 2021-05-08:
Yeah, this is one of those cases where the best I can say is "don't use macOS", macOS also sets archive bits on atmosphere/nintendo folders silently which breaks shit.

closed 2021-05-08 by SciresM

reneroth, 2021-05-08:
In case it came off wrong: I don’t need support in getting it to run using Mac OS, everything is working fine for me. I was merely proposing a way to prevent future users for running into that gotcha, if only via  a note in the readme :)

Skonikol, 2021-05-08:
You can use the free CleanMyDrive for example.

SciresM, 2021-05-08:
Yeah, windows/Linux don't create files that way.

reneroth, 2021-06-14:
Again, I do not need help in deleting hidden files or software recommendations, I've already solved this problem for me and even provided a solution in my first post. 

> Windows by itself doesn't create files on external drives by default. So it would happen only because user has screwed configuration or it's using some external app.
> And definitely it wouldn't create filename which first character is dot.

From my what I've gathered, not only Mac OS, but also Windows and some Linux Desktop distributions at least have the potential to create hidden files in some configurations. Those could easily be copied to the external drive by accident. Users might also get the update files from an aggregator that provides a readme.txt or something like that.  
I'm not sure how a discussion about the possibility of this problem occurring is productive when we're only here because it did, in fact, occur.

Daybreak already manually catches the `0x1a405` error code and provides a hint on giving a proper folder name (which is also user error). It might be helpful to catch the code referring to unexpected additional files and amend the error message to something more useful. It might also be helpful to create some kind of documentation on known issues.  
In the end, it's your project, your decision. Hopefully at least this issue will help some fellow users in search of an answer :)

masagrator, 2021-06-14:
> but also Windows and some Linux Desktop distributions at least have the potential to create hidden files in some configurations.

Officially windows doesn't create index files inside folders. It creates only one db file for folders with images that have created thumbnails. But update folder doesn't have images, so this point is not touching daybreak. If something is created inside folders except for this file, then this is not Windows fault.

reneroth, 2021-06-14:
I want to apologise if my suggestion has offended you or anyone else here. There is no need to be defensive, I never wanted to imply that there was something wrong with any part of Atmosphere or your OS of choice.  
If you refuse to document this unexpected behaviour that is absolutely your call to make and users with the same problem will hopefully stumble across this issue.

WinterMute, 2021-06-14:
> I want to apologise if my suggestion has offended you or anyone else here.

So, rather than continuing this passive aggressive approach where you submit an issue and apparently expect something to be done perhaps you could submit a PR to this open source project with the changes you'd like to see.

reneroth, 2021-06-14:
> > I want to apologise if my suggestion has offended you or anyone else here.
> 
> So, rather than continuing this passive aggressive approach where you submit an issue and apparently expect something to be done perhaps you could submit a PR to this open source project with the changes you'd like to see.

I am perfectly willing to do so if within my skills, but unsure which approach - if any - is the right one, hence this ticket and me asking which solution would be preferred. I am not a native speaker, so I probably worded some things in a way that made it seem like a support request.
I see three options:
- detecting when validation fails due to unexpected additional files and amending the error message
- creating an end-user-facing readme and collecting the FAQs and gotchas
- treating this issue as outside the scope of the project

chuynadamas, 2021-06-15:
> You can use the free CleanMyDrive for example.

This works like a charm with 12.0.3! :)

noirscape, 2021-06-15:
> So, rather than continuing this passive aggressive approach where you submit an issue and apparently expect something to be done perhaps you could submit a PR to this open source project with the changes you'd like to see.

This response is far more passive agressive than anything else being send in this thread. A user encountered a problem, they were told and explained the reason and the answer, they are trying to help find a way to fix it so that other users don't encounter it. They aren't even necessarily asking for a software change, just for this to be detected. I see no passive agressiveness here besides your comment.

As for potential solutions; wouldn't it be possible to make daybreak only scan for files with an nca/meta.nca extension in it's respective folders? That would be the cleanest and would help avoid any sort of random file generation due to systems trying to index folders/file managers trying to store extra information about a folder.

That would in my eyes be the cleanest way to avoid this from being a problem.

SciresM, 2021-06-15:
I don't actually think there's a problem here -- Daybreak is a strict reimplementation of the official gamecard update partition installation process, and fails on the same cases that official gamecard update installation would fail.

Keeping it this way is my preference.

Frankly, my advice is as before: don't use macOS, which already causes other issues with archive bits.

If you wanted to PR a special warning message in Daybreak when there are extra files, that would probably be accepted.

I also found the OP's comments to be passive aggressive, on the record, particularly in the context of their necro-ing a closed issue from over a month ago.