Atmosphere-NX/Atmosphere #1488: Patch for "control.nacp" on a specific title

OperationNT414C, 2021-05-08:
## Feature Request

### What feature are you suggesting?
#### Overview:

Being able to override some properties of "control.nacp" for a specific title

#### Smaller Details: 

Add an "ini" file in "/atmosphere/contents/titleID" which describes the list of settings that we wish to override (or maybe directly the raw "control.nacp" file but it will be less user-friendly)

#### Nature of Request:  

Addition

### What component do you feel this would best fit within?

Probably the same component as layered FS ?

### Why would this feature be useful?

A plugin could require that a specific service is enabled. For example, the "nx-btred" plugin (which allow to use Bluetooth headset) needs that the game activates "audrec:u" service. It can be done by having the video capture flag set to 0x02: https://switchbrew.org/wiki/NACP

closed 2021-05-08 by SciresM

OperationNT414C, 2021-05-08:
Ok, thank you for your feedback.
Indeed, it seems that video capture is enough for nx-btred.

cucholix, 2021-10-03:
Is there any reason why you are against ncapdm override?