Atmosphere-NX/Atmosphere #1495: Atmosphère 0.19.2 breaks with Update 12.0.2

CarlosRamirezGriego, 2021-05-12:
### What's the issue you encountered?

For the sake of science (Ok, it was by mistake), I updated the CFW console and the 12.0.2 patch was installed.
This broke the latest version of Atmosphere.

### How can the issue be reproduced?

Have installed 0.19.2.
Update the console to 12.0.2

### Crash Report

Couldnt get it

### System Firmware Version

12.0.2

Masamune3210, 2021-05-12:
Restore from a backup or wait, simple as that. Its mostly expected that new updates will break atmos

CarlosRamirezGriego, 2021-05-12:
> Restore from a backup or wait, simple as that. Its mostly expected that new updates will break atmos

Yeah, I have no rush, I will wait for the team to release a newer version. Just thought about warning other people with this bug xD

maheer461, 2021-05-12:
Thank you for the heads up :)

xdinko, 2021-05-12:
any update? i tried with hekate v5.5.6 & Nyx v1.0.3 but after boot go in error with
**Fatal error: [NXBOOT] Unable to identify package!**

WokerLOL, 2021-05-12:
So I was not able to run any games because I "had to" install the newest firmware. How would you go about installing it without removing AutoRCM and not bugging Atmosphere ofc?

SciresM, 2021-05-12:
Yeah this doesn't need a separate user issue.

closed 2021-05-12 by SciresM