Atmosphere-NX/Atmosphere #1496: Fatal error occurring constantly

Fudgecake4567, 2021-05-12:
**What's the issue you encountered?**

a fatal error occurred while running atmosphere, program id 0100000000000005, error desc "std::abort() called (0xffe)"

**How can the issue be reproduced?**

loading into Hekate and launching atmosphere from launch menu or fusee-primary

**Crash Report**

A fatal error occurred while running Atmosphere.
Program ID: 0100000000000005
Error Desc: std::abort() called (0xffe)
*generic log location*
Press POWER to reboot

[errorreport.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6466003/errorreport.zip)

**System Firmware Version**

not actually sure, it may have updated on its own but i don't know

WokerLOL, 2021-05-12:
So I sadly made the exact same mistake you did. You did auto update the system firmware while in Atmosphere. You now have to wait for the devs to make the CFW support the newest firmware. Do not boot anything now as there is a high chance you can brick your console.

Fudgecake4567, 2021-05-12:
okay, tell me if you ever get it working

WokerLOL, 2021-05-12:
Have you made a backup of your nand? If you did, launch into Hekate and restore it if you don't want to wait.

SciresM, 2021-05-12:
Re-open with the issue template actually fully filled out, it's missing information.

closed 2021-05-12 by SciresM