Atmosphere-NX/Atmosphere #1499: Constant Crash when loading an Owned Game

KaiserTialoc, 2021-05-12:
I Keep getting this crash when trying to open a copy of Pokemon Shield that i bought and downloaded from the eShop

I was playing the game until 20 minutes ago and then realised there was a system update and needed to update my atmosphere build aswell so i did all that and dowloaded the new fusee.bin

[image]

I am also using the ldn_mitm 1.8.0 on this switch if that make a difference and Pokemon Shield is on v.1.3.2

any help would be appreciated

nwhiteman42, 2021-05-12:
Reopen with the issue template filled out

SciresM, 2021-05-12:
Re-open with issue template filled out.

closed 2021-05-12 by SciresM