Atmosphere-NX/Atmosphere #1501: Constant Crash when trying to play game

KaiserTialoc, 2021-05-12:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

Console crashes when trying to play a game, the game is a legit owned copy of Pokemon Shield bought from the eShop last year
It was working completely fine earlier today.

i had been working on a shiny den to be used with the sysbot and i finally got the Shiny seed i wanted, that was when i perfomred the latest system update and then i updated atmoshphere and hekate
after the update when i launch Pokemon Shield, as soo as the save loads the console crashes

### How can the issue be reproduced?

Happens everytime i launch the game

### Crash Report

Atmosphère Crash Report (v1.5):
Mesosphere:                      Disabled
Result:                          0x290 (2144-0001)

Process Info:
    Process Name:                qlaunch
    Program ID:                  0100000000001000
    Process ID:                  0000000000000081
    Process Flags:               000010b7
    User Exception Address:      0000001ce5b77000
Exception Info:
    Type:                        User Break
    Address:                     000000559244ea44 (qlaunch.nss + 0x24ea44)
    Break Reason:                0x0
    Break Address:               00000007900fb868
    Break Size:                  0x4
Crashed Thread Info:
    Thread ID:                   000000000000020d
    Stack Region:                00000007900f8000-00000007900fc000
    Registers:
        X[00]:                   0000000000000000
        X[01]:                   00000007900fb868
        X[02]:                   0000000000000004
        X[03]:                   0000000000000008
        X[04]:                   0000000000000000
        X[05]:                   0000000000000000
        X[06]:                   0000000000000000
        X[07]:                   0000000000000000
        X[08]:                   0000000000000000
        X[09]:                   0000000000000000
        X[10]:                   0000000000000000
        X[11]:                   0000000000000000
        X[12]:                   0000000000000000
        X[13]:                   0000000000000000
        X[14]:                   0000000000000000
        X[15]:                   0000000000000000
        X[16]:                   0000000000000000
        X[17]:                   0000000000000000
        X[18]:                   0000000000000000
        X[19]:                   0000000000000000
        X[20]:                   0000000000000000
        X[21]:                   0000000000000000
        X[22]:                   0000000000000000
        X[23]:                   0000000000000000
        X[24]:                   0000000000000000
[... 318 lines cut ...]
        X[11]:                   0000000000000000
        X[12]:                   0000000000000000
        X[13]:                   0000000000000000
        X[14]:                   0000000000000000

### System Firmware Version

12.0.2/AMS M.19.3/S</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee
- Official release or unofficial build: 19.3.0
- Do you have additional kips or sysmodules you're loading: ldn_mitm 1.8.0
- Homebrew software installed: JKSV, Checkpoint (although not working), EdiZon, FTPD Pro 3.0.0, Daybreak, NXDumptool
- EmuMMC or SysNAND: SysNAND

### Additional context?

Image for context of the onscreen error

[image]

SciresM, 2021-05-12:
The crash report you pasted is unrelated to the crash you describe/took an image of.

ldn_mitm has some issues with latest atmosphere, remove ldn_mitm to stop getting the crash.

closed 2021-05-12 by SciresM

KaiserTialoc, 2021-05-12:
This is the crash report from the location the screen directs me to

apologies in advance however the file is in .bin format so i zipped it to be able to post it here

[report_00000000599c9a05.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6468935/report_00000000599c9a05.zip)