Atmosphere-NX/Atmosphere #1502: Game cart loading is slowed.

H4X0R46, 2021-05-12:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand.  newest 19.3 update on Mariko seems to recognize game carts slowly. Not a huge issue because it WORKS, but worth mentioning.]</br>
[ Did you make any changes related to Atmosphère itself? I have a couple sysmodules. lanplay, edizon, tesla, and edizon overlay. ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. Simply insert a game cart into a Mariko unit that's running 12.0.2 and atmosphere 0.19.3. ]</br>

### Crash Report

Doesn't cause a crash, super minor issue.</br>
]</br>
</br>

### System Firmware Version

12.0.2</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate and booting fusee-secondary.bin
- Official release or unofficial build: Official release 0.19.3
- Do you have additional kips or sysmodules you're loading: Mentioned in details above.
- Homebrew software installed: [Nothing major, some emulators and such, that's it.]
- EmuMMC or SysNAND: EmuMMC
  - [ file-based ]

### Additional context?

Masamune3210, 2021-05-12:
please actually fill out the issue template.....

SiliconMage, 2021-05-14:
I'm not the OP, but:

## Bug Report

### What's the issue you encountered?

A game from a cart takes an abnormally long time to load.  It takes about 3-5 minutes to get past the Nintendo Switch splash screen while a game is being loaded from the cart.

### How can the issue be reproduced?

1.) Installed Atmosphere 0.19.3 
2.) Install ARCropolis (https://github.com/Raytwo/ARCropolis)
3.) Install EdiZon (https://github.com/WerWolv/EdiZon)
4.) Install nx-ovlloader (https://github.com/WerWolv/nx-ovlloader/releases/latest)
5.) Install Tesla-Menu (https://github.com/WerWolv/Tesla-Menu/releases/latest)
6.) Install EdiZon Overlay (http://werwolv.net/downloads/EdiZonOverlay.zip)
7.) Install Smash Ultimate Training Modpack (https://github.com/jugeeya/UltimateTrainingModpack)
8.) Reboot to Payload
9.) Insert a valid copy of Super Smash Bros. Ultimate into the Switch.
10.) Attempt to launch the game and wait.

### Crash Report

This error doesn't cause the system to crash.

### System Firmware Version

12.0.2

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe
- Official release or unofficial build: Official
  - 0.19.3
- Do you have additional kips or sysmodules you're loading: Tesla-Menu, nx-ovlloader
- Homebrew software installed: EdiZon, Tesla-Menu, ARCropolis, EdiZon Overlay, Smash Ultimate Training Modpack
- EmuMMC or SysNAND: SysNAND

### Additional context?

This didn't happen immediately upon updating to 0.19.3 of Atmosphere.  It only started happening after I launched the game a few times.  

Other things that may be relevant:
1.) My SD card is formatted in Ex-FAT, and it is 128GB in size.
2.) I transferred my Super Smash Bros. Ultimate save from a previous Switch to this Switch.  I don't think this would cause an issue, but I know it breaks EdiZon, so I figured I would give it a mention.
3.) As a reminder, this didn't occur at first on 0.19.3 and I never had this issue with previous versions of Atmosphere.  

Thanks for everything!  I hope this helps!  Let me know if you need additional information!

nwhiteman42, 2021-05-15:
Have you tried with a clean installation?

ThatNintendoNerd, 2021-06-02:
ARCropolis attempts to contact a server at runtime for beta updates, which if down takes a while for it to timeout, i.e. minutes. This probably is that very case, and it'd be worth testing now or offline/in airplane mode to see if the problem is remedied. You can also modify `sd:/atmosphere/contents/01006A800016E000/romfs/arcropolis.toml` and set the provided IP address to 0.0.0.0 to prevent it from contacting any server.

It's likely not an Atmosphère problem if the above solutions work or if the game loads at normal speeds without ARCropolis.

SciresM, 2021-06-15:
Closing this because ^ seems like the likely answer, and I can't think any other mechanism. 

Contact me on discord if this reproduces using stock ams with no addons.

closed 2021-06-15 by SciresM