Atmosphere-NX/Atmosphere #1508: Panic Occurred

costelabr17, 2021-05-15:
[](url)Title ID: 524850534F53454D
ERROR: KRNL ( 0xF00 )

this error shows when i try boot atmosphere using fss0 as setup, if i use fusee-primary as payload, console only show black screen, 
sysnand 11.0.1 
[image]
[https://www.mediafire.com/file/otc90f6tkzxo84e/report_01d8a747.bin/file](url)

SciresM, 2021-05-15:
Screenshot of your atmosphere/contents folder?

The error occurs due to invalid INI binary in memory when kernel is doing setup.

costelabr17, 2021-05-15:
i put atmosphere 0.19.3 release without any sysmodule the inject fusee-primary now show my this error and generate this report [https://www.mediafire.com/file/e3jtj9tnvdd1lvf/report_0000000001d7260d.bin/file](url)
[image]

SciresM, 2021-05-15:
You did not post a screenshot of the atmosphere/contents folder, which you were requested to do.

Do you happen to have some kind of mod adding extra RAM or something crazy to your switch?

My best guess is that or that your installed package2 is malformed.

SciresM, 2021-05-15:
Didn't mean to close this yet.

costelabr17, 2021-05-15:
sorry i will send it on monday, i forgot my console at work, tks for the help

costelabr17, 2021-05-17:
[image]

Erionn, 2021-05-28:
I'm having a similar issue, but with a different error code.  I updated from 12.0.1 to 12.0.2 last night and Atmosphere 19.3, had no issues until I rebooted this morning and started getting this:

[image]

I've tried a fresh install of atmosphere, I have nothing else on the SD card other than what is necessary for it to boot. I can get into hekate just fine, OFW boots fine, but I get that same error any time I try to boot into atsmophere, whether it's through hekate or just through injecting fusee-primary.

SciresM, 2021-05-28:
@Erionn your problem is completely unrelated to the one this issue is discussing.

That said, create a new issue using the issue templates, and upload your report binary.

closed 2021-06-15 by SciresM