Atmosphere-NX/Atmosphere #1510: Failed to match warm boot with fuses - Sleep Mode won't work

Moriarty92x, 2021-05-15:
Im having some trouble after updating to the latest Atmosphere. I have a Mariko Switch.

I updated Atmosphere, Hekate and sigpatches to latest versions. Updated emunNAND to 12.0.1 with daybreak. Reset, everything worked fine. Later I updated sysNAND to 12.0.2 and now I keep getting this error when I try to boot emuNAND: "Failed to match warm boot with fuses - Sleep Mode won't work". I'd really appreciate some help.

Moriarty92x, 2021-05-16:
Sorry, will post with correct format.

## Bug Report

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>

I had both NANDs on 11.0.1. I updated Atmosphere, Hekate and sigpatches to latest versions. Updated emunNAND to 12.0.1 with DayBreak. Reset. emuNAND worked fine. Later I updated sysNAND to 12.0.2. Downloaded the latest MH Rise update overnight, downloaded SNES classics from newly bought online subscription, played MH for a bit. Rebooted the system and tried to boot emuNAND and now I keep getting this error: "Failed to match warm boot with fuses - Sleep Mode won't work. " emuNAND boots but can't enter sleep mode. I get a black screen after trying to wake and have to press the power button for 30 seconds to restart system. 

[ Did you make any changes related to Atmosphère itself? ]</br>
No.

### How can the issue be reproduced?
Whenever I boot emuNAND-

SciresM, 2021-05-19:
You need to boot the version corresponding to your real fuse count with atmosphere in order to be able to boot a version that doesn't match, on mariko.

Figure out how many fuses you have burnt via hekate, then boot the corresponding version. Sleep will be fixed after this.

closed 2021-05-19 by SciresM