Atmosphere-NX/Atmosphere #1512: Error Code: 2002-3502 (0x1b5c02)

yaelboy36, 2021-05-16:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>

I was playing The Legend of Zelda when suddenly the switch crashed. It shows me this error.

[ Did you make any changes related to Atmosphère itself? ]</br>

No, my emunand is on 11.0.1 and everytime I enter it always crashes.

[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

https://gist.github.com/yaelboy36/4e19b26a13ba311c3bf17d8440f9e2d1

### System Firmware Version

12.0.2 SysNAND, 11.0.1 EmuNAND</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

I was using atmosphere fusee primary, but I tried also whit hekate and stills not working.
Atmosphere Official release 0.19.3

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: Official release 0.19.3
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND: EmuMMC - Partition-based
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

yaelboy36, 2021-05-16:
[image]

shaneshanekavanagh, 2021-05-18:
Having a similar issue myself. I'm on 10.2.0 cfw Emu, and 12.0.2 ofw. It's a Mariko switch with an sxos chip. ofw works perfect.

[image]

yaelboy36, 2021-05-19:
Have you figured out what to do?

yaelboy36, 2021-05-20:
I couldn't find a solution so I make a backup of my saves with TegraExplorer, I found a tutorial. Then I remade my emunand and now I'm installing all my games again, but the error finally disappeared

SciresM, 2021-05-20:
The error 2002-3502 indicates a failure to access mmc. My best guess would be that there's some corruption in your SD card file emummc, or something.

closed 2021-06-15 by SciresM

Willhulle, 2021-07-20:
I had the same problem.
This error occurs due to damage to an internal component of the Nintendo switch. The SD Card FCP. A Specialized assistance can make this repair, and this component you can find for sale on the internet. After replacing this part, i had no more problems with the console. Search for a video on YouTube called "Nintendo switch SD Card FCP replacement", and you Will see the problem being solved.

Masamune3210, 2021-07-20:
....unless I'm wrong that's just the connector on the main board for the SD reader daughter board. Should probably work easiest to hardest and make sure that issue is a issue before just telling people that they need something

Mart189, 2021-09-05:
> I couldn't find a solution so I make a backup of my saves with TegraExplorer, I found a tutorial. Then I remade my emunand and now I'm installing all my games again, but the error finally disappeared

Can you link me the guide you used to backup the saves with Tegraexplorer, I can't find anything and I need to remake the emunand

BennyAlex, 2021-11-08:
Hello, i also have this error. looks like my emunand got corrupted somehow.
Before a fresh install from my sysnand, is there a way to backup the saves in emunand? Since I cant boot into the system, I cant use checkpoint, so I need some tool to get them with PC or from RCM.

yaelboy36, 2021-11-08:
> Hello, i also have this error. looks like my emunand got corrupted somehow. Before a fresh install from my sysnand, is there a way to backup the saves in emunand? Since I cant boot into the system, I cant use checkpoint, so I need some tool to get them with PC or from RCM.

This is the tutorial: https://suchmememanyskill.github.io/guides/switchextractsaves/

BennyAlex, 2021-11-08:
Funny, I deleted all files from sd exept the nand and copied clean atmosphere and hekate files on it. Then I booted into atmosphere using fuse.bin emunand, same error appeard.
Then I tried to turn it off or to boot back into RCM, but it just restarted again into atmosphere. After doing this like 3 or 4 times, it booted in switch home screen.  I have no clue why it works now, but I will probably never turn it off again, hopefully it doesnt crash :D

Mart189, 2021-11-08:
> Funny, I deleted all files from sd exept the nand and copied clean atmosphere and hekate files on it. Then I booted into atmosphere using fuse.bin emunand, same error appeard. Then I tried to turn [...]

Trust me, it will crash again the next time you reboot your Switch. Your emummc got corrupted.
Now that you are inside Atmosphere, backup all your saves with JKSV or Checkpoint, transfer them to your pc, and format and test your sdcard folllowing this guide: https://3ds.eiphax.tech/sd.html
Then partition the sdcard with tegraexplorer and remake EmuMMC with Hekate following this guide: https://nh-server.github.io/switch-guide/user_guide/emummc/partitioning_sd/
and then you can reinstall latest Atmosphere.

P.S. if you're using MacOS to prepare your sd card, please use Windows instead, cause that was the reason error 2002-3502 happened to me (supposedly).

Masamune3210, 2021-11-09:
The main reason macOS is bad for messing with switch stuff is it doesn't know how to keep its hands off of random files and has a really bad habit of setting the archive bit on things which confuses the hell out of the switch and causes a myriad of issues

leeberlin, 2023-05-11:
Any one can fix this error ? i have same and dont know how to fix it!

trino7, 2024-12-10:
He probado reinstalando emunand,juegos,con otro tarjeta sd y con otro lector de micro sd y el error sigue,ahora no tengo ninguna duda de que el problema es del conector de la placa donde conecta el lector de micro sd.

Masamune3210, 2024-12-27:
Yeah, the port on the board for the SD card reader is really sensitive to damage; it's not really designed for multiple insertions.

smobanan, 2024-12-28:
> Yeah, the port on the board for the SD card reader is really sensitive to damage; it's not really designed for multiple insertions.

Im getting the same error. Tried reformatting my sd card several times, some other stuff I read online, doing a visual inspection of the port, but it looks perfectly fine. How would I even go about finding out if its actually damaged?

osakilo, 2025-01-04:
[image]

It was working fine originally, but when I opened the game the next day, it turned out like this. Half of the games on the SD card show an error message and cannot run, while some games run normally. 
Even I delete the games that show this error and reinstall them, they still display the same message (the installation has no errors, but they just cannot run). 
I checked the memory card with software and found no bad sectors. I'm really at a loss now

DavidDubinin, 2025-08-10:
> Im getting the same error. Tried reformatting my sd card several times, some other stuff I read online, doing a visual inspection of the port, but it looks perfectly fine. How would I even go about [...]

Just a quick update, turns out nothing was broken- It was just a poorly seated sd card reader. Removed it, plugged it back in gently, making sure to give it a good push and it's been just fine since. GL to y'all figuring this out