Atmosphere-NX/Atmosphere #1515: Game cartridge not being read properly.

NotOnii, 2021-05-17:
# The issue:
I updated to Nintendo Switch firmware 12.0.2. I reinstalled all the new Atmosphere files and Hekate 5.5.6 & Nyx.
I booted up with fusee-primary and unlocked my Switch. After I got to the home menu, an error came up, saying it couldn't read my game cartridge anymore (Skyrim NX Edition). It was working fine just prior to updating everything, so I believe this has something to do with Atmosphere's update (0.19.3).
I've also tested some other game carts, and the same issue came up. Even rebooting several times didn't help.

Other than the game cartridge not being read properly, there's no issue.

Maybe this is an issue only I'm facing, but I hope it's not permanent.

kevdog-png, 2021-05-17:
same thing for me, works in stock fw

SciresM, 2021-05-17:
This isn't an issue. Set nogc = 0 in config.

you also didn't use the issue templates.

closed 2021-05-17 by SciresM