Atmosphere-NX/Atmosphere #1516: [ERROR] Fatal error: [NXBOOT] Failed to determine mariko warmboot firmware

shurillu, 2021-05-17:
## Bug Report

### What's the issue you encountered?

I'm unable to load the EmuMMC version 11.0.1. The error message is:

`[ERROR] Fatal error: [NXBOOT] Failed to determine mariko warmboot firmware`

### How can the issue be reproduced?

- Mariko Switch
- Original firmware version 12.0.2 (HOS)
- Fresh install of Atmosphere (latest, 0.19.3)
- fusee-primary.bin from Atmosphere 0.19.3
- Fresh install of Hekate (latest, 5.5.6)
- SXGear v. 1.1
- an EmuMMC with 11.0.1 HOS

I tried to boot the EmuMMC using:
- Hekate payload and then the fusee-primary.bin
- fusee-primary.bin as payload

In both cases, I get the first Atmosphere splash screen (the bluish one with stars) and suddenly  a black screen with the error message:

`[ERROR] Fatal error: [NXBOOT] Failed to determine mariko warmboot firmware`

### Crash Report

ND

### System Firmware Version

Original firmware version 12.0.2
EmuMMC firmware version 11.0.1

### Environment?

- Bootloader: hekate 5.5.6 and fusee-primary (0.19.3) / fusee-primary (0.19.3) directly
- Official release of Atmosphere 0.19.3
- Do you have additional kips or sysmodules you're loading: No, only fresh install
- Homebrew software installed: No, only fresh install
- EmuMMC 11.0.1 file-based 

### Additional context?

I've booted the Mariko Switch with original firmware and updated to the 12.0.2 (from 12.0.1) and I forgot to update the EmuMMC before. 
With the original firmware 12.0.1 I was able to boot the EmuMMC with 11.0.1 HOS.
I can still boot the original firmware but if I try to boot the EmuMMC I got the error listed.
I boot the original firmware without hekate (I use the old SX OS function "genuine boot") 
I also tried to make a new uSD with all fresh installs (hekate and Atmosphere) and dumping a new EmuMMC from the original firmware - a fresh EmuMMC with 12.0.2 HOS from the original firmware - and then booting with this EmuMMC with success.
So I replaced the emuMMC folder (the one with 12.0.2 HOS) with the old one (the one with 11.0.1 HOS) but I got the same error listed.

Thank you for all your hard work.

Stefano

SciresM, 2021-05-19:
You need to boot the version corresponding to your real fuse count with atmosphere in order to be able to boot a version that doesn't match, on mariko.

Figure out how many fuses you have burnt via hekate, then boot the corresponding version. Sleep will be fixed after this.

closed 2021-05-19 by SciresM

bonafp, 2021-06-12:
I have the same problem, How to boot the corresponding version of the real fuse count. With hekate, I've got 15-0. Please help. Thanks.