Atmosphere-NX/Atmosphere #1517: Error when going into sleep mode

osolzinho, 2021-05-21:
## Bug Report

</br>
</br>

### What's the issue you encountered?

[I changed to atmosphere from SXOS and noticed every time I leave switch on sleep mode it was showing this error. ]</br>
</br>

### How can the issue be reproduced?

[ * ]</br>
[ Press the power button to enter sleep mode. ]</br>

### Crash Report

[https://gist.github.com/osolzinho/a4b74bc481fef519e0203f15e317c30a ]</br>

### System Firmware Version

12.0.2</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee-primary.bin
- Official release or unofficial build: Official 0.19.3
- Do you have additional kips or sysmodules you're loading: I don't know
- Homebrew software installed: [ N/A ]
- SysNAND

[image]

emveezz, 2021-06-03:
If I remember well I had "same" error last time (same kind of screen) and went to the process of reformatting everything.

CFW is working if you reboot manually? On my side it was no more working at all after this.

My SD card was in exFAT, I formatted it to FAT32 after googling this issue.

SciresM, 2021-06-15:
This is an "invalid wlan device" error, there's nothing ams does that can cause this.

Closing the issue because it's likely hardware problems, if the error does not reproduce on stock firmware let me know.

closed 2021-06-15 by SciresM