Atmosphere-NX/Atmosphere #1519: "The Software was closed because an error occurred."

Electric1447, 2021-05-22:
## Bug Report

N/A

### What's the issue you encountered?

Getting "The Software was closed because an error occurred." on some Doom64EX and Cannonball

### How can the issue be reproduced?

Opening Doom64EX or Cannonball from hbmenu (non-applet or applet mode)

### System Firmware Version

12.0.2

### Environment?

- Bootloader: fusèe
- Official release version 0.19.3
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Doom64EX
- EmuMMC partition based

### Additional context?

Both Doom64EX and Cannonball were set up with their needed files.

SciresM, 2021-05-22:
This seems *extremely* unlikely to be an atmosphere problem. Upload crash reports?

Electric1447, 2021-05-22:
I have recompiled cannonball myself and it worked.
Prob a dkp or libnx issues

closed 2021-05-22 by Electric1447

fincs, 2021-05-22:
May I add, older versions of libnx had several issues that became a problem later on in later versions of Horizon OS. If the homebrew binary in question is sufficiently old, then yes, a recompile with latest tools and libraries provided by devkitPro will solve these issues.