Atmosphere-NX/Atmosphere #1520: "The software was closed because an error occurred" when trying to launch any homebrew

SagarcSaga, 2021-05-25:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ When i try to run a homebrew from the album i get the error "The software was closed because an error occurred"]</br>
[ Did you make any changes related to Atmosphère itself? No]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

12.0.2</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee primary
- Official release or unofficial build: 
  - [ Official release version 0.19.3 (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue: I switched from SXOS to Atmosphere because SXOS doesn't support FW above 11.0.0. On SXOS the homebrews worked perfectly (tested on FW 11.0.0 for SXOS and Atmosphere to confirm if the problem is on the CFW or the homebrew itself). ]

SciresM, 2021-05-25:
You're using latest release NROs launched via latest hbmenu? Is this exclusive to application mode/applet mode?

SciresM, 2021-05-25:
Could you also provide an example of a failing homebrew? 

Stuff compiled before 2019 needs recompile to work on 12.x/ams 0.19

SagarcSaga, 2021-05-25:
> Could you also provide an example of a failing homebrew?
> 
> Stuff compiled before 2019 needs recompile to work on 12.x/ams 0.19

I found out the problem. My homebrews were outdated. I downloaded the latest version and now it worked. Thank you very much ;D

closed 2021-05-25 by SagarcSaga