Atmosphere-NX/Atmosphere #1522: Appstore stoped working

420kingof420, 2021-05-27:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version 12.0.2
[image]

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]
[image]

420kingof420, 2021-05-27:
I'm sorry I didn't fill it out right but I don't understand it I'm very new sorry

SciresM, 2021-05-27:
1) issue template not filled out correctly

2) homebrew compiled before 2019 needs recompile to work with firmware 12.x/ams 0.19.

closed 2021-05-27 by SciresM

420kingof420, 2021-05-27:
Sorry mate  but the homebrew is 2020

SciresM, 2021-05-27:
It's a homebrew issue, not an atmosphere issue, man.

420kingof420, 2021-05-27:
Ok thanks do you have any idea of what I should do

SciresM, 2021-05-27:
Get the developer to update/fix the homebrew.

420kingof420, 2021-05-27:
So don't downgrade the firmware to a working one that compatible

SciresM, 2021-05-27:
That will make games not work once games require higher firmware.

The above wasn't a joke, but anyway this isn't a help forum, it's for bugs.