Atmosphere-NX/Atmosphere #1524: Random „Panic occurred while running Atmosphere” with title ID: 010000000000002B

the3ye, 2021-05-28:
## Bug Report

Random „Panic occurred while running Atmosphere” with title ID: 010000000000002B 

### What's the issue you encountered?

Random behaviour. First, Home Button stops working as intended while in games or Home Screen — it does not take me back to Home Screen. Holding button works, though — can use Quick Settings. To fix this, I have to reboot console — then I’m getting this error. After hitting Power to continue, Switch boots normally into Atmosphere and is working for a while. After some time (few hours of playing) it’s happening again. I followed https://nh-server.github.io/switch-guide/ to start hekate/Atmosphere from scratch but it’s still happening after that. Plus the random 2105-0602 crash happened few minutes ago. On my side, I edited BCT.ini with "nogc = 0”, system_settings.ini with "dmnt_cheats_enabled_by_default = u8!0x0" and exosphere.ini with "blank_prodinfo_emummc=1".

### How can the issue be reproduced?

Random, can’t really provide steps to reproduce it. It happens in games, home screen and in sleep mode.

### Crash Report

Crash report is linked to 2105-0602 error that happened around 15-20 minutes after „Panic occurred while running Atmosphere” with title ID: 010000000000002B.

https://gist.github.com/the3ye/8bbc5df6772ad024a5b8d83b087bcb4b

bin file from fatal_errors folder from Panic Error screen:

https://wetransfer.com/downloads/43ea8894606a0ce6f34f9ef770579bb220210528074614/f6438eec1d23b46a0d2c4b53038af0e920210528074626/c5bed5

### System Firmware Version

12.0.2</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - official Hekate 5.5.6 chainloading fusee-primary.bin for Atmosphère 0.19.3
 
- EmuMMC or SysNAND:
  - EmuMMC partition-based

SciresM, 2021-05-28:
What custom sysmodules do you have?

please post a screenshot of your atmosphere/contents directory.

the3ye, 2021-05-28:
> What custom sysmodules do you have?
> 
> please post a screenshot of your atmosphere/contents directory.

I’m not using sysmodules and don’t have _contents_ in atmosphere folder

SciresM, 2021-05-28:
You have a corrupt PRODINFO/system save.

Follow the instructions/directions in #1256 for guidance. You'll need to restore a non-corrupt backup and/or delete the system settings savedata.

closed 2021-05-28 by SciresM

the3ye, 2021-05-28:
Thanks for helping but I have one last question. Do I have to restore whole clean backup of NAND and start from scratch or can I try and restore only PRODINFO/system save files?

SciresM, 2021-05-28:
You can certainly try lesser restores. If issues persist, try greater ones.

leopitto92, 2022-07-21:
Did you solve it?
I'm experiencing the same issue...

SciresM, 2022-07-21:
@leopitto92 do *not* comment on closed issues like this.

If you must, make a new issue, but really you should seek support on discord.