Atmosphere-NX/Atmosphere #1525: [ERROR] Fantal error: [NXBOOT] Key derivation failed! - after NAND Restore

DevCoud18, 2021-05-28:
## Bug Report

### What's the issue you encountered?

An boot error after doing an NAND Restore.
I tryed:
- Redoing the NAND restore with hekate. The Backup was 100% clean, created with hekate.
- Building an new NAND with the current official methods like choidujour or tegraxplorer script.
- Updating Atmosphere and hekate to his latest versions. Nothing changes.
- Booted Atmosphere via fuse (Error above) and hekate. (Hekate gives an pkg1 decryption error on CFW and pkg2 decryption error on OFW)

### System Firmware Version
4.1.0 (efuse count: 5)

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Booted Atmosphere via fusee (Error above) and hekate. (Hekate gives an pkg1 decryption error on CFW and pkg2 decryption error on OFW)

- Official release or unofficial build: official builds
(Atmosphere v0.19.3 and hekate v5.5.6)
 
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Only HBMenu (.nro-files) no nsp's or other similar content
- EmuMMC or SysNAND: SysNAND (SysMMC)

### Additional context?

- Additional info about your environment:
An very hard brick... hope i have an chance to get it fixed

Picture:
[image]

SciresM, 2021-05-29:
Sounds like you're not putting your keyblobs into BCT. Just flash a clean nand for 12.x or whatever which doesn't use console unique keyblobs in BCT.

closed 2021-05-29 by SciresM

DevCoud18, 2021-05-29:
And how do i do that? I followed every method example with the tegraxplorer script but i don't know where or with witch tool i can make a nand without the keyblobs or restoting an nand with the correct keyblobs inside BCT. Have you an idea for an tool, writen guide or similar that i can try to fix this issue. Thx anyway for your reply