Atmosphere-NX/Atmosphere #1528: DNS_Mitm crashes when redirecting with minecraft

fennectech, 2021-05-31:
## Bug Report

### What's the issue you encountered?

Using DNS_MITM to redirect minecraft servers results in panic

Stock atmosphere   no sysmodules

### How can the issue be reproduced?

Use DNSMITM with a dns config such as this 

# Nintendo telemetry servers
127.000.000.001 receive-%.er.srv.nintendo.net
127.000.000.001 receive-%.dg.srv.nintendo.net
# Nintendo Update Server
127.000.000.001 sun.hac.lp1.d4c.nintendo.net
# Minecraft DNS Addresses
209.182.219.167 geo.hivebedrock.network
209.182.219.167 hivebedrock.network
209.182.219.167 mco.mineplex.com
209.182.219.167 play.inpvp.net
209.182.219.167 mco.lbsg.net
209.182.219.167 mco.cubecraft.net
209.182.219.167 play.galaxite.net

### Crash Report
There is no crash report only a fatal error 
http://transfer.sh/1ye5rRu/report_4c9a1b2b9be.bin
http://transfer.sh/a87J9/report_0703cde68.bin
### System Firmware Version

12.0.2
### Environment?

-Hekate
- Official release or unofficial build: 
  - [ Official release version 0.19.3 (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
  Deepsea (but also happens on bone stock atmosphere)
- Do you have additional kips or sysmodules you're loading:None
- Homebrew software installed: [ Lots installed but none running ]
- SysNAND:

fennectech, 2021-05-31:
Disabling these redirections and using a dns server to perform them makes everything work fine
so its not to do with the servers being redirected

SciresM, 2021-05-31:
Run with dns.mitm logging enabled, and post the relevant logs?

SciresM, 2021-05-31:
Also, the two reports you posted are completely unrelated to eachother.

SciresM, 2021-05-31:
One of the reports you posted is "SD card data crc error", which is to say a hardware communications error when talking to your SD card...

fennectech, 2021-05-31:
```DNS Mitm:
Adding defaults to redirection list.
Selecting hosts file...
Skipping /hosts/sysmmc.txt because it does not exist...
Selected /hosts/default.txt
Redirections:
    `play.galaxite.net` -> 209.182.219.167
    `mco.cubecraft.net` -> 209.182.219.167
    `mco.lbsg.net` -> 209.182.219.167
    `play.inpvp.net` -> 209.182.219.167
    `mco.mineplex.com` -> 209.182.219.167
    `hivebedrock.network` -> 209.182.219.167
    `geo.hivebedrock.network` -> 209.182.219.167
    `sun.hac.lp1.d4c.nintendo.net` -> 127.0.0.1
    `receive-lp1.dg.srv.nintendo.net` -> 127.0.0.1
    `receive-lp1.er.srv.nintendo.net` -> 127.0.0.1```

fennectech, 2021-05-31:
the debug log is a 0 byte file

SciresM, 2021-05-31:
Please delete all fatal errors, generate a report using stock atmosphere (not deepsea), and post that report.

Given one of the reports you uploaded is an sd card hardware error, I'm inclined to distrust the reports you uploaded insofar as their relation to the issue.

fennectech, 2021-05-31:
One of them may be an older report    i thought i deleted them tho

fennectech, 2021-05-31:
http://transfer.sh/17lb17t/report_594281669.bin  Brand new one   only one on the file system

SciresM, 2021-05-31:
"svc::ResultOutOfSessions()" on attempt to connect to sfdnsres's mitm port.

This doesn't really make any sense, naturally -- the mitm port's session limit is identical to the non-mitm port's limit...

fennectech, 2021-05-31:
Somethings wonkey    The dns_mitm debug log seems to have its contents changing   sometimes its empty and sometimes its P??   Im going to reinstall atmosphere after doign a disk check   shits getting weird.

SciresM, 2021-05-31:
That's just the file not being flushed.

fennectech, 2021-05-31:
i think that report might be something else   as im getting a fatal now slightly before and that might have been this    Ether way   somethings weird  ill send a report if i can get one that i KNOW is it (when showing the menu that querys the servers)

SciresM, 2021-05-31:
You're using zero additional custom system modules, correct? Screenshot of /atmosphere/contents to verify this?

fennectech, 2021-05-31:
Ive got lots of stuff in there but its currently all disabled.   Im recreating it now with nothing enabled on a fresh boot

SciresM, 2021-05-31:
"disabled" not how that works. 

If stuff is in that folder, assume it's on. I asked you to trigger the issue without custom sysmodules, please.

fennectech, 2021-05-31:
Ive now renamed my entire atmosphere folder and greated a fresh one and copied my configs.  Now ill recreate it.

fennectech, 2021-05-31:
http://transfer.sh/1HMJqFS/report_056ca0a04.bin

fennectech, 2021-05-31:
https://gyazo.com/2e83684d2a20d8e1bc952882631e04c9  No contents folder as shown here

fennectech, 2021-05-31:
https://pastebin.com/nT6tWDup  Here is the content of debug.log

fennectech, 2021-05-31:
and startup.log
https://pastebin.com/gf1a46zU

SciresM, 2021-05-31:
Hm. None of the domains you mentioned are actually referenced in the log.

Out of sessions is a very strange error, here.

fennectech, 2021-05-31:
Thats even weirder.  It did crash slightly before i was able to bring up the menu
but with this issue im unable to recreate the issue :/   Im not able to get into the screen where it querys the servers.  I figured the game is quierying them before   But  this could be related to the issue in some roundabout way.

SciresM, 2021-05-31:
Ah, resolver library creates a new session for each query, and it's multithreaded.

That's kind of horrific.

I believe I see the problem, potentially, though.

I'll have a test fix kip in ~10 minutes.

I'll upload it and then you'll place it inside /atmosphere/kips, and then *boot using fusee-primary, not hekate* to verify, yes?

fennectech, 2021-05-31:
got it  Can i chainload through hekate or do i need to upload it manualy

SciresM, 2021-05-31:
Use fusee-primary directly, to minimize the possibility of error.

fennectech, 2021-05-31:
oi   Ill have to program my shitty payload launcher lol

SciresM, 2021-05-31:
Please test using the following: http://transfer.sh/1fC9czE/sm.kip 

It has abort conditions edited to be slightly more lenient, in a way that I guess this might be triggering.

fennectech, 2021-05-31:
i cant find my blue one so ill have to use the SX gear payload

fennectech, 2021-05-31:
I got a "failed to initalize the filesystem io error "  but it cointinued loading when i hit the power button
 Then it fataled durring boot   Im pretty sure this is related to the sx   ill try via heakate cuz i got no other payload sender

SciresM, 2021-05-31:
I don't really understand why you can't just e.g. reboot to RCM from hekate, then use a USB cable connected to your PC like normal.

fennectech, 2021-05-31:
I dont use a PC to send my payload  i use an SX gear

SciresM, 2021-05-31:
That's obnoxious. Whatever, chainload fusee-primary through hekate.

fennectech, 2021-05-31:
http://transfer.sh/1rBgySu/report_02b5f9dbb.bin
  Fatals almost imediately on opening the gmae

SciresM, 2021-05-31:
That report is "SD card tuning failed", i.e. a hardware SD card issue.

Please re-test, it's unrelated to the issue we're diagnosing.

fennectech, 2021-05-31:
http://transfer.sh/1DgGGa8/report_05148f676.bin

fennectech, 2021-05-31:
What tool are you using to open those reports   id like to look into a few of mine too

SciresM, 2021-05-31:
The report you just uploaded is *not* using the sm kip I uploaded.

Please post a screenshot of your /atmosphere/kips folder, with the sm kip I sent in it.

You're booting via fusee-primary, *not hekate*, yes?

And I'm using a hex editor.

fennectech, 2021-05-31:
https://gyazo.com/e72d182a84caee9ce2b421e49dbc818b

SciresM, 2021-05-31:
It's also possible you uploaded an outdated report, somehow, from before you added the kip I sent, I guess.

I'll be unavailable for the next few hours starting in ~5 minutes.

SciresM, 2021-05-31:
Okay, then I'd guess you booted via hekate instead of fusee, or you uploaded an old report. 

Either way, the sm I sent wasn't loaded, which means error on your end.

fennectech, 2021-05-31:
Its not fataling this time    I dunno why (it did error on the first try booting (unable to init sd card) but its working now

SciresM, 2021-05-31:
I do not like this error, but okay.

closed 2021-05-31 by SciresM

fennectech, 2021-05-31:
its likely the SD card    I have a PNY 1TB sd card cuz i have so many games

fennectech, 2021-05-31:
(and no   none of them are pirated    Most are shovelware from the eshop   i love to buy the supeer cheap 50 cent games and see if any of them are any good   ive found a few games worth their prices

fennectech, 2021-05-31:
Would you be willing to make atmosphere more fault tolerant with SD cards.  The rest of the OS seems more fault tolerant and so does hekate.      With these SD cards that interact poorly with the OS becoming more and more common fault tolerance (automatic retry on failure and aborting on repeated failures) would allow the system to work better with these error prone (but still fully functional) cards.

SciresM, 2021-05-31:
this isn't an atmosphere issue. The error is being generated by FS sysmodule.

"make atmosphere more fault tolerant" isn't even a thing, if atmosphere is receiving an error it means the OS has *already* tried repeatedly and gotten failures repeatedly.

Get a less shit SD card, and this isn't related to the issue.

commit 8fea8d9b (2021-05-31): sm: lighten abort restrictions on mitm handle acquisition (closes #1528)