Atmosphere-NX/Atmosphere #1531: Compilation failed fatal error: switch/services/htcs.h: No such file or directory

fennectech, 2021-06-01:
Im trying to compile atmosphere today.  Ive got the latest atmosphere libraries and propperly did git submodule init and update.    Its bailing here  

`/home/fennectech/switchdev/Atmosphere/libraries/libstratosphere/source/htcs/client/htcs_session.cpp:21:10: fatal error: switch/services/htcs.h: No such file or directory
   21 | #include <switch/services/htcs.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [/opt/devkitpro/devkitA64/base_rules:14: htcs_session.o] Error 1
make[2]: *** [Makefile:92: lib/libstratosphere.a] Error 2
make[2]: Leaving directory '/home/fennectech/switchdev/Atmosphere/libraries/libstratosphere'
make[1]: *** [Makefile:8: libstratosphere] Error 2
make[1]: Leaving directory '/home/fennectech/switchdev/Atmosphere/libraries'
make: *** [Makefile:42: libraries] Error 2

`

I cant find the hcts.h file in the repo or on github

fennectech, 2021-06-01:
im thinking this is probally my screwup but im unsure what ive screwed up.

SciresM, 2021-06-01:
atmosphere requires libnx-master.

closed 2021-06-01 by SciresM

fennectech, 2021-06-01:
Alright. Thanks.