Atmosphere-NX/Atmosphere #1532: Fatal error after booting "oxffe"

StanMOI, 2021-06-01:
### Bug Report

### What's the issue you encountered?

After updating to Atmosphere 0.19.3 I get error code "oxffe" just after booting, just after Atmosphere logo
[image]

[image]

I have updated myself files and my USB key with the fusee primary.bin file
I followed the usual procedure and it always worked before but this time i have a Black Screen.

### How can the issue be reproduced?

After updating and launching via RMC Loader

### Crash Report

[report_0000000007e1c737.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6576106/report_0000000007e1c737.zip)

### How can the issue be reproduced?

After Booting via RMCLoader.

### System Firmware Version

12.0.2</br>

### Environment?

Atmosphere official

### Additional context?

I use RMCLoader that I updated to last **payload.bin** (renamed fusee-primary.bin to payload.bin)
[image]

[image]

Contents folder have those folders, but even after delating all of them, the error are still here :
[image]

Here is content inside my SD card:
[image]

On my SD card I have old folders:
"atmosphere-old" (I rename the old folder cause I had Black screen problem after updating to 0.19.3
"emuMMC" (I dont use it), cause I launching  directly Atmosphere via "RMC Loader"

Here is content inside Atmosphere folder:
[image]

Sigpatches was updated form here  https://github.com/ITotalJustice/patches/releases
With "Fuse" file

Thanks for help

Masamune3210, 2021-06-01:
Piracy is not supported

StanMOI, 2021-06-01:
> Piracy is not supported

What in my post are « Piracy »?

fincs, 2021-06-01:
You literally linked to sigcheck patches, which are exclusively used for piracy.

closed 2021-06-01 by fincs

StanMOI, 2021-06-01:
> You literally linked to sigcheck patches, which are exclusively used for piracy.

why you close every time !
I corrected my post

masagrator, 2021-06-01:
> > You literally linked to sigcheck patches, which are exclusively used for piracy.
> 
> why you close every time !
> I corrected my post

Because hiding evidence won't change anything...