Atmosphere-NX/Atmosphere #1534: Titles will not load, "Checking if game can be played"

Mewthree3, 2021-06-01:
## Bug Report

### What's the issue you encountered?

I have the latest sigpatches, running hekate 5.5.6 and atmosphere 0.19.3.
No Nintendo account was on the system (had one before, but I system formatted)
The Nintendo account that used to be on it is on another system (it seems that the other system is designated as the secondary).
I had installed some dumps of games that I legally own, that were on my eShop account.

I'll try to go on the Nintendo website and see what's up with the accounts.
Additionally, I'll also try to go on the other Switch, delete the user, and then put the user on the system again, to see if that solves the problem for me.

### How can the issue be reproduced?

Have Nintendo account on your hacked device.
Dump legally-owned games (from eShop).
Delete Nintendo Account from system.
Format hacked system (do not add a Nintendo Account afterwards), and install games.
Notice that the games will not load, and says "Checking if game can be played".

### System Firmware Version

12.0.2

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official Hekate, v5.5.6
- Do you have additional kips or sysmodules you're loading: Not AFAIK
- Homebrew software installed: No NSPs installed
- EmuMMC or SysNAND:
SysNAND

### Additional context?
**This is not a piracy issue, nor an issue with software SPECIFICALLY DESIGNED for piracy**
- Additional info about your environment:
Latest Atmosphere and SP running

fincs, 2021-06-01:
[image]

Piracy is not supported.

closed 2021-06-01 by fincs

SciresM, 2021-06-01:
The things you describe are piracy, on the record.