Atmosphere-NX/Atmosphere #1538: Feature request: Disable firmware updates

gjimenezf, 2021-06-05:
Please implement an option in atmosphere to disable firmware updates.
I use internet with the default block of telemetry using DNS mitm of atmosphere, and I like being able to update games using Nintendo servers but I want to disable only firmware updates from Nintendo because that usually breaks atmosphere due to incompatibility.
I have seen many people have this problem and 
they ask you for fast updates of atmosphere version because his child updated switch firmware by mistake.
Having this option as default in atmosphere to disable switch firmware updates from Nintendo servers would be a great addition to atmosphere.

shchmue, 2021-06-05:
this is likely out of scope anyway but i can only imagine the telemetry this would produce when facilitating the game update would be easy ban bait

SciresM, 2021-06-05:
No, sorry.

atmosphere wants users to be on the latest firmware update -- adding a thing that makes people usually not be on the latest firmware update would make my life significantly more complicated as a developer.

also, you did not use the issue templates.

closed 2021-06-05 by SciresM