Atmosphere-NX/Atmosphere #1539: Fatal Error DABRT (0x101)

kennethdz, 2021-06-06:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

I cannot play Burnout: Paradise Remastered or Need for Speed Hot Pursuit, error occurs before main menu. </br>
Recently updated to 0.19.3 to update my firmware to 11.0.1 to play these games. </br>
I also recently reformatted my SD card from exFAT to FAT32 and created an android partition using Hekate </br>

### How can the issue be reproduced?

[ * ]</br>
Only happens when I try to play those two games</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>

https://gist.github.com/kennethdz/1b7ed8f93895235a3cf6c7d6f083eb26

[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

11.0.1 AMS 0.19.3</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate 
- Official release or unofficial build: official 5.5.6
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: sigpatches are on the SD card as well.
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND: SysNAND
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2021-06-06:
The report you uploaded is not the report relevant to the crash you mentioned.

Please upload the relevant fatal error report binary, the path to the file is mentioned when the error occurs.

kennethdz, 2021-06-06:
https://github.com/kennethdz/Fatal-Error-Binary

kennethdz, 2021-06-06:
https://github.com/kennethdz/Atmosphere-Error-Reporting

SciresM, 2021-06-15:
This is an error in atmosphere 0.19.3 regarding nullptr hostnames in dns.mitm that was resolved for 0.19.4.

closed 2021-06-15 by SciresM

kennethdz, 2021-06-15:
Just updated to 0.19.4, but I am still unable to play the game. Here is the most recent error binary. 

https://github.com/kennethdz/Atmosphere-Error-Reporting/blob/main/report_2ba141a0.bin

SciresM, 2021-06-15:
Looks like an error in the wildcardcmp function when processing empty string.

Pushed a fix for it to master.