Atmosphere-NX/Atmosphere #1541: A fatal error while loading atmosphere ID:010041544D530000

JoeKawasakii, 2021-06-09:
## Bug Report

### What's the issue you encountered?

Back in December I was using an nds emulator for switch on cfw and it crashed. Since then I could never get the cfw to load without it crashing and I haven't figured out what the problem is

### How can the issue be reproduced?

Boot to RCM
Launch Hekate
Boot into CFW

### Crash Report

[report_064387d3.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6621192/report_064387d3.zip)

### System Firmware Version

12.0.3

### Environment?

hekate 5.5.7

SciresM, 2021-06-09:
`settings::ResultSettingsNameEmpty()`

Your system_settings.ini is corrupted/malformed, fix it to not contain empty settings names.

closed 2021-06-09 by SciresM

Borrajo, 2021-06-09:
Run "archive bit fix" in hekate to fix it.