Atmosphere-NX/Atmosphere #1542: I accidentally updated my switch to fw version 12.0.3

kyrtos, 2021-06-09:
[report_0000000009ef082c.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6623509/report_0000000009ef082c.zip)
I'm using atmosphere 19.4, i tays "A fatal error has occured when running Atmosphere" the program id is 0100000000000004. I will attach the .bin error report

Mewthree3, 2021-06-09:
I'm using atmosphere 19.4 and have the same issue as well. (I thought 19.4 would cover the latest FW)

SciresM, 2021-06-09:
1. You did not use the issue templates. For further correspondence, re-open using the issue templates.

2. This is not an atmosphere issue, you have some outdated custom system module (probably mission control). Delete custom sysmodules and everything should work fine, re-install latest versions of custom sysmodules one by one to verify all is well with whatever you have.

closed 2021-06-09 by SciresM