Atmosphere-NX/Atmosphere #1544: fatal error occurred when running atmosphere title id 0100000000000004 after updating to firmware 12.0.3 with Daybrake

Dreas31, 2021-06-15:
## Fatak error occurred

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
I'm using Sturdust to launch Atmosphere and I was trying to update my EMUMMC partition from firmware 12.0.2 to the latest 12.0.3 but since I updated to 12.0.2 some homebrew apps won't lauch (Checkpoint, ChoiDujourNX, Edizon, etc) so I follow the update installation with Daybrake ([here](https://gbatemp.net/threads/atmosphere-error-when-trying-to-run-anything-from-album-error-code-2144-0001-0x290.588949/)), and everything seems to install OK since the reboot, now every time I try to launch Atmosphere I get this error:

[image]

I try to copy the latest atmosphere release on my SD, and doing the same for the Stardust but still happen de issue

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>

[report_00000000182854df.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6658570/report_00000000182854df.zip)

[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

12.0.3</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Sturdust with RCMLoader
- Official release or unofficial build: Official
  - [ Official release version 0.19.4 (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:N/A
- Homebrew software installed: [ * ]N/A
- EmuMMC or SysNAND: EmuMMC
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ] exFAT Partition

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Dreas31, 2021-06-15:
I also try to install the [sigpatches ](https://github.com/ITotalJustice/patches/releases) even but, the same result.

I don't know if will help, but in my Atmosphere/Content folder don't see any folder called 0100000000000004, and before the firmware update I reinstall a Theme on my emuMMC.

This is my Atmosphere folder

[image]

SciresM, 2021-06-15:
Update mission control, not an atmosphere issue.

closed 2021-06-15 by SciresM

Dreas31, 2021-06-15:
First of all thank you for your early answer, the problem was solved as you said, it was my mission control outdate. Thank you.

Could I ask you why the Checkpoint and ChoiDujourNX apps still don't work 

[image]

Sorry for the inconvenients and thank you so much again!

SciresM, 2021-06-15:
this is not a support forum, it's for atmosphere bugs.

that said, choidujournx doesn't work any more, use daybreak instead.

certain homebrew needs update since 12.0.0, contact the developers.