Atmosphere-NX/Atmosphere #1545: Random Fatal Report from error applet

Bappsack, 2021-06-15:
## Bug Report

Just uploading it here so it won't forgetten (already known via Discord).

Atmosphere randomly crashes while generating a error report.

This is a similar (or even the same issue) when the 'erpt_reports' Folder hits a certain amount of reports(16k~).

### What's the issue you encountered?

Fatal Error while generating a error report, Sword and Shield always generate a error report if your trade partner quits the Trade. 

### How can the issue be reproduced?

Let Atmoshpere generate a bunch of error reports. Trading in Swsh for example.

### Crash Report

[01623593938_0100000000001005.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6658886/01623593938_0100000000001005.zip)

### System Firmware Version

12.0.3</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusée
- Official release or unofficial build:  latest release.
- Do you have additional kips or sysmodules you're loading: sys-botbase, LDN-Mitm
- Homebrew software installed: none.
- EmuMMC or SysNAND: SysNAND

### Additional context?

- Additional info about your environment: FAT32 SD Card

[01623593938_0100000000001005.log](https://github.com/Atmosphere-NX/Atmosphere/files/6658873/01623593938_0100000000001005.log)

SciresM, 2021-06-15:
fwiw, this is just the error applet fataling because erpt has run out of memory, probably because of too many error reports.

To confirm: the issue only occurs with huge numbers of error reports?

closed 2021-06-15 by SciresM

Bappsack, 2021-06-15:
> To confirm: the issue only occurs with huge numbers of error reports?

Yes, it was around 16k files, now its like 500+/- files?