Atmosphere-NX/Atmosphere #1546: Add Breakpoints and Debugging to dmnt cheats

ReclaimerShawn, 2021-06-22:
## Feature Request

Adding breakpoints/debugging addresses and functions to dmnt cheats.

### What feature are you suggesting?
#### Overview:
Implement the ability for breakpoints/debugging to be set on addresses and functions. This would allow a user using an outside Homebrew software (like Edizon or Noexs) to debug an address, returning the functions that write to the address or even setting breakpoints on functions, freezing the game whenever they're called.

### Why would this feature be useful?
Would allow for people to make ASM Cheats for the Nintendo Switch without using programs like IDA Pro, which cost thousands of dollars, or Ghidra, which some people are paranoid of due to the NSA developing it. This would also make it significantly easier to make ASM cheats, as the functions would be tied to their addresses instead of needing to be searched for in a game's main.nso.

masagrator, 2021-06-22:
This is what tma2 is targetting with way more functions.

https://github.com/Atmosphere-NX/Atmosphere/pull/1405

masagrator, 2021-06-22:
Breakpoints and debugging are parts of tma2 reimplementation.

ReclaimerShawn, 2021-06-22:
I actually searched issues beforehand to see if this was there, and there was one duplicate that could no longer be commented on. Nice to know that's in the Pull Requests and was started more recently (2 months ago). I can't wait until that is implemented: it'd make my life a lot easier.

closed 2021-06-28 by SciresM