Atmosphere-NX/Atmosphere #1547: Tantei Bokumetsu LayeredFS issues

InochiPM, 2021-06-25:
## Bug Report

### What's the issue you encountered?

I'm trying to edit files for Tantei Bokumetsu, and when i add them to the romFS folder inside of atmosphere the changes do not reflect in game, as i think the game itself has a round-about way of pointing to the romfs folder, but im not fully sure thats the issue. 

### How can the issue be reproduced?
By making edits to any of the files inside the games romFS folder. 
for instance i edited the text in two files in the database folder and none of them reflected in the game.

### Crash Report
N/A

### System Firmware Version

12.0.3</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Hekate

SciresM, 2021-06-26:
Update executable for Tantei Bokumetsu (0100cba014014800) just does nn::fs::MountRom(), this will be covered by LayeredFS. There is nothing abnormal about how the game points to romfs.

What makes you think that LayeredFS isn't working vs your not editing files correctly?

Please post screenshots of your relevant layeredfs directories.

SciresM, 2021-06-28:
Closing because I don't think this is real.

closed 2021-06-28 by SciresM

InochiPM, 2021-06-28:
[image]