Atmosphere-NX/Atmosphere #1548: TABE-O-JA LayeredFS error

InochiPM, 2021-06-26:
## Bug Report

### What's the issue you encountered?
I'm also trying to translate TABE-O-JA into english but the text files are all in .json format and easily editable. 
Problem is the game itself does not reflect any sort of edits made to it with the text, as it's a unity game i do not know how LayeredFS works with them. 

### How can the issue be reproduced?
Dragging any one of the translated .json files into the correct folder in the layeredFS directory.

12.0.3</br>

### Environment?

Hekate 
Official release
SysNAND

### Additional context?
The game is built in Unity and uses the files it has to point the way to the streaming assets but im guessing that Atmos doesnt support these kinds of games with LayeredFS.

SciresM, 2021-06-26:
There is no known issue with unity games, unity/not unity will make no difference to layeredfs.

I'll look at it.

this and the other issue should have been a single issue.

SciresM, 2021-06-28:
This game also just calls nn::fs::MountRom().

I am pretty confident it's a user-not-making-game-mod-correctly problem, and not an atmosphere problem.

closed 2021-06-28 by SciresM

InochiPM, 2021-06-28:
[image]
The folder structure is correct atmosphere make the file for it in the contents folder. 
It just refuses to work.