Atmosphere-NX/Atmosphere #1549: [ERROR] Fatal error: [NXBOOT] Couldn't parse boot0: !

Shabekuri, 2021-06-26:
## Bug Report

### What's the issue you encountered?

When i try to boot into the emuMMC i get the Error: Couln't parse boot0: !
Created like 4-5 Times a new emuMMC and Repaired my boot0 because i thought it may be faulty but the Error still persist.

### System Firmware Version

12.0.3

### Environment?

I'm using Hekate v5.5.7 to Launch Atmosphere v0.19.4.
I do not use any kips or sysmodules.
emuMMC is File-Based. Partition just give's me a Blackscreen.

### Additional context?

My Switch is a Patched Erista.
Here's a Screenshot of the Error: [Picture](https://media.discordapp.net/attachments/646630046731075600/858119052726501376/image0.jpg) (Hosted on Discord)

Regards

SciresM, 2021-06-28:
Sounds like you have a corrupt boot0. Restore a good backup?

closed 2021-06-28 by SciresM

Shabekuri, 2021-08-03:
Sorry for the late reply.
when i boot through the forbidden OS it works just fine. this only occurs when i boot straight in hekate and from there to atmosphere.

SciresM, 2021-08-03:
"Sounds like you have a corrupt boot0. Restore a good backup?"