Atmosphere-NX/Atmosphere #1555: Ability to disable game updates

GRAnimated, 2021-07-06:
## Feature Request

### What feature are you suggesting?
#### Overview:
For games where modding the latest version is harder, an option to disable the current update would be nice.
Many mods for games like Super Mario Odyssey are for 1.0.0 and uninstalling the update doesn't work if you have a cartridge due to a bug with Goldleaf.
My preferred method for launching the base game would be to put a file in the game's content folder, so mods can dictate whether you need 1.0.0 for them.

#### Smaller Details: 
N/A

#### Nature of Request:  
N/A

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>

### Why would this feature be useful?
This feature would fit in nicely with exefs/romfs mods to allow compatibility with some mods stuck on 1.0.0 for one reason or another.

SciresM, 2021-07-06:
I don't know how one would do this, and I don't think the feature is desirable in master...if you implement this and have suggestions for how it could be integrated in a configurable way, I'd be happy to think about it?

closed 2021-07-06 by SciresM

Ammako, 2021-07-06:
[DBI](https://github.com/rashevskyv/dbi) for resetting launch version on cartridge