Atmosphere-NX/Atmosphere #1556: Unable to identify package1

LordDurus, 2021-07-06:
## Bug Report

### What's the issue you encountered?
Unable to identify package1

Switch system update to 12.1.0 then updated to Atmosphère to 0.19.5
Yes.  Copied the files for 0.19.5 to the SD card.

### How can the issue be reproduced?
Boot the Switch in "recovery" mode
  -> Joycon jig + Vol + Power
Inject the Payload with TegraRcmGUI
 -> Payload file: fusee-primary.bin (0.19.5 version)
-------------
Boot into hekate
Select Launch
Select fusèe

[ Include a detailed step by step process for recreating your issue. ]

### Crash Report
I do not have current crash the most recent on is from 2021-05-26

### System Firmware Version

12.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
-> hekate:  Starts but will not launch Atmosphère; get's the Unable to identify package1
-> fusèe (my perferred way):  get's the Unable to identify package1
 Official release version 0.19.5
- Do you have additional kips or sysmodules you're loading:
   -> No, but I do use custom boot logo.
- Homebrew software installed: [ * ]
   -> Tinfoil
   -> FTPD
- EmuMMC or SysNAND: SysNAND

### Additional context?
  Switch boots normal (no joycon jig)
  I tried updating the the keys via lockpick.  Didn't think that would help, but worth a shot.
  I also installed the latest sig patch.
  
- Additional info about your environment: N/A

SquishyMp4, 2021-07-06:
you need to also update hekate via this link: https://github.com/CTCaer/hekate/releases

SciresM, 2021-07-06:
Not an atmosphere issue.

closed 2021-07-06 by SciresM

LordDurus, 2021-07-06:
I updated hekate as well.  But will try again and post back with how that works out.

LordDurus, 2021-07-06:
I updated hekate  again no good.  

I also get the error when booting with atmosphere bin file (fusee-primary.bin)

SciresM, 2021-07-06:
The issue is almost certainly that you haven't actually updated atmosphere.

LordDurus, 2021-07-06:
I'll try deleting atmosphere entirely and installing fresh

LordDurus, 2021-07-06:
Deleting atmosphere  did not work.  But re-downloading does it it to boot.

I should have done that in the first place.  Thank you both

EmmanDB2, 2022-01-31:
> 

How exactly did you fix the issue I've been trying everything but nothing seems to work

basement-guy, 2024-10-12:
I'm still having the same problem, even with using multiple different atmosphere versions