Atmosphere-NX/Atmosphere #1557: Can't jump in Game Builder Garage interactive lessons

gingerbeardman, 2021-07-06:
## Bug Report

### What's the issue you encountered?

Pressing B does not jump in Game Builder Garage interactive lessons. 

### How can the issue be reproduced?

1. Start Game Builder Garage for the first time
2. choose the only option available: interactive lessons
3. the first stage requires you to move with L and jump with B

### Crash Report

N/A

### System Firmware Version

12.0.3

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:

hekate

- Official release or unofficial build: 
  - official 0.19.4

- Do you have additional kips or sysmodules you're loading:

N/A

- Homebrew software installed:

N/A

### Additional context?

- Additional info about your environment:

I've tried joycons and also switch Pro controller. B doesn't jump using either.

SciresM, 2021-07-06:
Does this issue occur only when using atmosphere...?

gingerbeardman, 2021-07-06:
Yes

SciresM, 2021-07-06:
Please post a screenshot of /atmosphere/contents.

if the game works in non-ams, play it under non-ams in the meantime, though.

gingerbeardman, 2021-07-06:
Of course!

I'll upload a screenshot shortly.

gingerbeardman, 2021-07-06:
[image]

SciresM, 2021-07-06:
Hm. Okay, this is super weird. 

I cannot think of a single mechanism by which normal atmosphere would interfere in controller stuff, without a contents folder.

I'll ask to see if anyone can reproduce this, since I think people have used game builder garage fine previously(?)

SciresM, 2021-07-06:
This is reportedly part of the game. 

Try spamming it for a bit?

closed 2021-07-06 by SciresM

SciresM, 2021-07-06:
Confirmed by watching a let's play that this not working is literally part of the game.

Discord user uploaded this: 
[image]

What did you mean when you said it worked on non-atmosphere?

gingerbeardman, 2021-07-06:
OMG. 

Thanks!

gingerbeardman, 2021-07-06:
Regarding your final question, my kids have been playing it on their stock Switch with no problems.

Totally my fault with the dodgy reporting in this instance. Apologies!