Atmosphere-NX/Atmosphere #1560: Crash when starting game download for app store.

LordDurus, 2021-07-09:
## Bug Report

### What's the issue you encountered?
Crash when starting any downloaded (app store) or when starting some carts.
Not all carts crash, but at least 1 does.
Some homebrew apps crash, but not all.
   
### How can the issue be reproduced?
Start a downloaded (app store) game.
Start Mystery of Solaris DX (Homebrew)

### Crash Report
[01625844452_010000000000001f.log](https://github.com/Atmosphere-NX/Atmosphere/files/6792549/01625844452_010000000000001f.log)
[01625844448_0100000000000033.log](https://github.com/Atmosphere-NX/Atmosphere/files/6792550/01625844448_0100000000000033.log)

### System Firmware Version
12.1.0

### Environment?
- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: 
  normally fusèe tried hekate same crash.
- Official release or unofficial build: Official  0.19.5

- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: 
  Checkpoint
  Daybreak
  Goldleaf
  HB App Store
  Mystery of Solaris DX
  NX Themes (no theme actually applied)
  NX-Shell
  nxdumptool
  RetroArch
  sys-cll manager
  Tinfoil
- EmuMMC or SysNAND: SysNAND

### Additional context?
When booting normal all carts and app store games work.
Screen shot of the crash
[image]

- Additional info about your environment:
Atmosphère: 0.19.5
hekate: v5.5.8 & Nyx v1.0.5
Patches: OS 12.1.0 and AMS 0.19.5

fincs, 2021-07-09:
> Goldleaf
> Tinfoil

Piracy is not supported.

closed 2021-07-09 by fincs

LordDurus, 2021-07-09:
I use Tinfoil to copy my carts to the SD card, but thanks for assuming

fincs, 2021-07-09:
That still qualifies as piracy.

LordDurus, 2021-07-09:
That can't be right then ripping a CD you bought to your MP3 player would also be piracy.

But anyways this would fall under "Fair Use"; as long it only for use by the owner for the cart.  Well in the U.S anyways.